magnusja/libaums #247: ArrayIndexOutOfBoundsException when reading FatFile found in ClusterChain.kt:157

cybex-dev, 2020-02-15:
Hi

Came across a `ArrayIndexOutOfBoundsException` - the specifics around the exception eludes me at the moment as I saw this in a Firebase crash report, I will try and update this with all the information I can once I am able to replicate the issue, but for the moment I want to post this here and bring it to your attention.

### Problem
Could not read valid USB file correctly

### Expected behavior
Read valid USBFile in its entirety

### Actual behavior
Failed to read file, exact cause unknown

#### Stacktrace of Exception
> Fatal Exception: java.lang.ArrayIndexOutOfBoundsException: length=1; index=1
>        at com.github.mjdev.libaums.fs.fat32.ClusterChain.read$libaums_release(ClusterChain.kt:157)
>        at com.github.mjdev.libaums.fs.fat32.FatFile.read(FatFile.kt:104)
>        at com.github.mjdev.libaums.fs.UsbFileInputStream.read(UsbFileInputStream.kt:79)
>        at com.company.example.libs.FATUtils.copyFrom(FATUtils.java:107)
>        at com.company.example.libs.FATUtils.copyFrom(FATUtils.java:59)
>        at com.company.example.activites.LoadingActivity$1.onMassStorageReady(LoadingActivity.java:113)
>        at com.company.example.libs.StorageUtils.onMassStorageReady(StorageUtils.java:314)
>        at com.company.example.libs.StorageUtils.getUsableUSBMassStorageDevice(StorageUtils.java:133)
>        at com.company.example.activites.LoadingActivity.lambda$onCreate$0$LoadingActivity(LoadingActivity.java:169)
>        at com.company.example.activites.-$$Lambda$LoadingActivity$Kc3ikVBqpggN1SSjCNjFbwHs4Jk.run(:2)
>        at java.lang.Thread.run(Thread.java:919)

### Code where problem occurs
```
// Get file and check if correct file based on path
UsbFile usbFile= cd(srcFileSystem.getRootDirectory(), srcUsbFilePath);
if (!usbFile.getAbsolutePath().equals(srcUsbFilePath)) {
        // Incorrect file, handle appropriately
	Log.e(TAG, "copyFrom: File " + srcUsbFilePath + " does not exist");
	if (copyListener != null)
		copyListener.onCopyError("File " + srcUsbFilePath + " does not exist");
	return FileUtils.FileStatus.NotFound;
}

// Proceed to read file
//...
InputStream inputStream = new UsbFileInputStream(usbFile);
try {
	long fileLength = usbFile.getLength();
	while ((count = inputStream.read(bytes)) != -1) {          **// Offending line @ FATUtils.java:107 - see Stacktrace**
		out.write(bytes, 0, count);
		total += count;
		if (copyListener != null)
			copyListener.onCopyProgress(total, fileLength);
	}
} catch (IOException e) {
	e.printStackTrace();
	Log.e(TAG, "copyFrom: An IO Error occurred while copying file");
	if (copyListener != null)
		copyListener.onCopyError("An IO Error occurred while copying file");
	return FileUtils.FileStatus.IOError;
}
```

magnusja, 2020-02-19:
Hey there,

can you maybe link the firebase report? Does it has anything to do with this library?

I remember that someone already reported this a while back but I thought we fixed this ..

kuza2010, 2020-02-25:
Hey there

In one part my application during read operation I get an issue like that above.

````
JNI DETECTED ERROR IN APPLICATION: JNI NewByteArray called with pending exception java.lang.ArrayIndexOutOfBoundsException: length=1; index=1
2020-02-19 12:40:00.080 31459-31642/com.my.app A/zygote64: java_vm_ext.cc:504] at void com.github.mjdev.libaums.fs.fat32.ClusterChain.read$libaums_release(long, java.nio.ByteBuffer) (ClusterChain.kt:157)
2020-02-19 12:40:00.080 31459-31642/com.my.app A/zygote64: java_vm_ext.cc:504] at void com.github.mjdev.libaums.fs.fat32.FatFile.read(long, java.nio.ByteBuffer) (FatFile.kt:104)
2020-02-19 12:40:00.080 31459-31642/com.my.app A/zygote64: java_vm_ext.cc:504] at int com.sandisk.everest.fs.libaums.UsbInputStream.read(byte[], int, int) (UsbInputStream.java:38)
````

magnusja, 2020-02-25:
Hmm that is bad. Can you log what with what offset/length you are calling the read method?

cybex-dev, 2020-02-25:
hi @magnusja 

_this feels like a useless comment but I will say it in anycase._

I think my issue appeared when attempting to read an incomplete file. (I copied over a file but it did not complete, which was the file I was trying to read). I doubt this makes any difference, I have not yet come across this issue again as the original problem occurred during development. If I find anything else, I will post an update.

kuza2010, 2020-03-01:
hi @magnusja 

So, I investigate what went wrong in my case. The issue appears after https://github.com/magnusja/libaums/issues/237
In one part of my app I used a secret store to keep some files in encrypted. It seems, that some system information uncompleted for the secret store when it is closing (https://github.com/magnusja/libaums/issues/237 appear during copying process). And after that, when I try to open the secret store I got the ArrayIndexOutOfBoundsException.

If I find anything that can help, I will post an update.

magnusja, 2020-03-03:
Hey,

so this is then a follow up error of the first error. If writing is unsuccessful then the cluster chain has no end indicator.

cybex-dev, 2020-05-23:
@magnusja Hi Magnus

So it seems I have gotten a number of additional crashes, however these have a different output to that shown above.

Instead of 

> Fatal Exception: java.lang.ArrayIndexOutOfBoundsException: length=1; index=1

I show a number of crashes with the following Stack Trace

> Fatal Exception: java.lang.ArrayIndexOutOfBoundsException: length=0; index=0
>        at com.github.mjdev.libaums.fs.fat32.ClusterChain.read$libaums_release(ClusterChain.kt:157)
>        at com.github.mjdev.libaums.fs.fat32.FatFile.read(FatFile.kt:104)
>        at com.github.mjdev.libaums.fs.UsbFileInputStream.read(UsbFileInputStream.kt:79)

I have added additional logging to try and get some information. Would any additional info from my side assist you in resolving this problem?

magnusja, 2020-05-24:
Well this crash is probably because of an interrupted write to a cluster chain. What do you think is the best way to handle this? Check the length and return zero bytes?

magnusja, 2023-03-11:
Fixed with v0.9.4

closed 2023-03-11 by magnusja