magnusja/libaums #248: compile error on version 0.7.5

pkuoliver, 2020-03-23:
### Problem
Compile Error
i want compile by myself, but it cannot be complied, Looks like some files are missing。

### Expected behavior

### Actual behavior

#### Stacktrace of Excpetion
CMake Error at CMakeLists.txt:11 (add_library):
Cannot find source file:
null/libusb/core.c
Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
.hxx .in .txx

### Code where problem occurs
add_library( libusb
             SHARED
             ${LIBUSB_SRC}/core.c
             ${LIBUSB_SRC}/descriptor.c
             ${LIBUSB_SRC}/hotplug.c
             ${LIBUSB_SRC}/io.c
             ${LIBUSB_SRC}/sync.c
             ${LIBUSB_SRC}/os/linux_usbfs.c
             ${LIBUSB_SRC}/os/poll_posix.c
             ${LIBUSB_SRC}/os/threads_posix.c
             ${LIBUSB_SRC}/os/linux_netlink.c)

magnusja, 2020-03-23:
Have a look here
https://github.com/magnusja/libaums/blob/develop/libusbcommunication/README.md#compile-yourself

closed 2020-03-23 by magnusja

pkuoliver, 2020-03-23:
sorry to miss this tips.

another complile error:

[10/10] Linking C shared library ..\..\..\..\build\intermediates\cmake\debug\obj\armeabi-v7a\liblibusb.so
FAILED: cmd.exe /C "cd . && D:\Android\SDK\ndk-bundle\toolchains\llvm\prebuilt\windows-x86_64\bin\clang.exe  --target=armv7-none-linux-androideabi16 --gcc-toolchain=D:/Android/SDK/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64 --sysroot=D:/Android/SDK/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -fno-addrsig -march=armv7-a -mthumb -Wa,--noexecstack -Wformat -Werror=format-security  -O0 -fno-limit-debug-info  -Wl,--exclude-libs,libgcc.a -Wl,--exclude-libs,libatomic.a -static-libstdc++ -Wl,--build-id -Wl,--warn-shared-textrel -Wl,--fatal-warnings -Wl,--exclude-libs,libunwind.a -Wl,--no-undefined -Qunused-arguments -Wl,-z,noexecstack -shared -Wl,-soname,liblibusb.so -o ..\..\..\..\build\intermediates\cmake\debug\obj\armeabi-v7a\liblibusb.so CMakeFiles/libusb.dir/D_/libusb-master/libusb/core.c.o CMakeFiles/libusb.dir/D_/libusb-master/libusb/descriptor.c.o CMakeFiles/libusb.dir/D_/libusb-master/libusb/hotplug.c.o CMakeFiles/libusb.dir/D_/libusb-master/libusb/io.c.o CMakeFiles/libusb.dir/D_/libusb-master/libusb/sync.c.o CMakeFiles/libusb.dir/D_/libusb-master/libusb/os/linux_usbfs.c.o CMakeFiles/libusb.dir/D_/libusb-master/libusb/os/poll_posix.c.o CMakeFiles/libusb.dir/D_/libusb-master/libusb/os/threads_posix.c.o CMakeFiles/libusb.dir/D_/libusb-master/libusb/os/linux_netlink.c.o  -llog -latomic -lm && cd ."
D:\libusb-master\libusb/core.c:2271: error: undefined reference to 'usbi_clock_gettime'

D:\libusb-master\libusb/core.c:2603: error: undefined reference to 'usbi_clock_gettime'

D:\libusb-master\libusb/io.c:2631: error: undefined reference to 'usbi_clock_gettime'

D:\libusb-master\libusb/io.c:1213: error: undefined reference to 'usbi_clock_gettime'

clang: error: linker command failed with exit code 1 (use -v to see invocation)

ninja: build stopped: subcommand failed.

pkuoliver, 2020-03-23:
sorry, my mistake.
it worked.