magnusja/libaums #251: Play fail and no host name

CronZhu, 2020-04-07:
start code : 
                                HttpServer server = new AsyncHttpServer(8000);
                                fileServer = new UsbFileHttpServer(file.getParent(), server);
                                Log.d(TAG," baseUrl : "+fileServer.getBaseUrl());
                                try {
                                    fileServer.start();
                                } catch (IOException e) {
                                    e.printStackTrace();
                                }
play code:  
                             MediaPlayer player=new MediaPlayer();
 try {
     player.setDataSource(MyApplication.getContext(),Uri.parse(fileServer.getBaseUrl()+file.getName()));
 ...   
  player.prepare();
     player.start();
 } catch (IOException e) {
     e.printStackTrace();
 }

it print :  
baseUrl : http://:8000/   
W/MediaPlayer: Couldn't open http://:8000/media223.mp4: java.io.FileNotFoundException: No content provider: http://:8000/media223.mp4
I dont know why  no host name and it wrongs when play，Can give me some suggestion ？Thanks

CronZhu, 2020-04-07:
now host name is right,but still print warn log : 
W/MediaPlayer: Couldn't open http://localhost:8000/media223.mp4: java.io.FileNotFoundException: No content provider: http://localhost:8000/media223.mp4

magnusja, 2020-04-07:
You might want to try this https://github.com/google/ExoPlayer

Other than that a complete example would be great which I can just open and run. Just looking at this I am not sure what is going on...

CronZhu, 2020-04-08:
> You might want to try this https://github.com/google/ExoPlayer
> 
> Other than that a complete example would be great which I can just open and run. Just looking at this I am not sure what is going on...

I try to run this project but it wrongs when compile :
debug|arm64-v8a :-- Detecting CXX compiler ABI info
debug|arm64-v8a :-- Detecting CXX compiler ABI info - done
debug|arm64-v8a :-- Detecting CXX compile features
debug|arm64-v8a :-- Detecting CXX compile features - done
debug|arm64-v8a :Using libusb path null
debug|arm64-v8a :-- Configuring done
debug|arm64-v8a :CMake Error at CMakeLists.txt:11 (add_library):
debug|arm64-v8a :  Cannot find source file:
debug|arm64-v8a :    null/libusb/core.c
debug|arm64-v8a :  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
debug|arm64-v8a :  .hxx .in .txx
debug|arm64-v8a :CMake Error: CMake can not determine linker language for target: libusb
debug|arm64-v8a :-- Generating done
debug|arm64-v8a :-- Build files have been written to: D:/Android/Projects/libaums-develop/libusbcommunication/.cxx/cmake/debug/arm64-v8a
executing external native build for cmake D:\Android\Projects\libaums-develop\libusbcommunication\CMakeLists.txt

FAILURE: Build failed with an exception.

magnusja, 2020-04-08:
If you want to build libaums including libusbcommunication yourself then please refer to this manual: https://github.com/magnusja/libaums/tree/develop/libusbcommunication#compile-yourself

CronZhu, 2020-04-09:
Thank you，I run the project successfully. But when I play the video ，it would copy before playing. How can I play video without copying ? Some file explorer app such as ES APP，can play the video without copying.And seemingly they dont use exo-player .

magnusja, 2020-04-09:
Well what you did in the first post should work. Other than that there are hundreds of ways to play a video file. In theory you just read the file from the disk and then put the bytes you just read into some video player.

CronZhu, 2020-04-10:
Now I can play the video with MediaPlayer .Thanks

closed 2020-04-10 by CronZhu

hugonnyy, 2022-11-16:
@CronZhu hello, would u please tell me how do you play video with mediaPlayer? I get the same problem.