magnusja/libaums #252: UsbMassStorageDevice.init

cherryzhang-zq, 2020-04-13:
Hello, when I call "UsbMassStorageDevice.init()",I will get a notification -- USB removed unexpectedly

So,can you help me?

cherryzhang-zq, 2020-04-13:
Notes: It also can still read and write.
I just don't want USB removed

magnusja, 2020-04-13:
This is an Android generated notification libaums can nothing do about.

If you use libaums and the USB drive is mounted within Android, Android looses cannot access it anymore and therefore shows this notification. But this is perfectly fine.

closed 2020-04-13 by magnusja

cherryzhang-zq, 2020-04-14:
Thank you! I get it.
But it will make the USB mass storage error. The U disk inserted in the computer is abnormal and needs to be repaired.
You would still care, wouldn't you?

magnusja, 2020-04-14:
Well this has nothing to do with the notification you are getting I would assume. What does the repair log say? (Please open another issue specifically for that)