magnusja/libaums #253: Error transferring command; errno 0 null

smoore6, 2020-05-08:
### Problem

Cannot write more than 100-ish files to the drive in a row, and drive loses connection. This is only happening on new drives that we are using, and cannot be reproduced on all drives. I have tried reformatting the drive and it does not help. 

### Expected behavior

To be able to continuously write files to the drive

### Actual behavior

receive error when attempting to write the drive, and drive loses connection

#### Stacktrace of Excpetion

D/ClusterChain: Init a cluster chain, reading from FAT
D/ClusterChain: Finished init of a cluster chain
D/ClusterChain: grow chain
D/FsInfoStructure: writing to device
I/FAT: allocating clusters finished
E/ScsiBlockDevice: Error transferring command; errno 0 null
D/ScsiBlockDevice: Reset bulk-only mass storage
W/ScsiBlockDevice: sending bulk only mass storage request
D/ScsiBlockDevice: Trying to clear halt on both endpoints
W/AndroidUsbCommunication: Clearing halt on endpoint UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0] (direction 128)
2020-05-08 18:01:46.013 16002-16816/
W/AndroidUsbCommunication: Clearing halt on endpoint UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0] (direction 0)
E/ScsiBlockDevice: Error transferring command; errno 0 null
D/ScsiBlockDevice: Trying to reset the device
D/AndroidUsbCommunication: Performing native reset
W/AndroidUsbCommunication: ioctl failed! errno 0 null
W/AndroidUsbCommunication: USB device will likely require new discovery and permissions

I also see this when I run first aid on the drive following the issue:

/Backups/Nexus 6P/Camera/10H.jpg has too many clusters allocated (logical=0, physical=2195456)

### Code where problem occurs

```
val buf = ByteArray(fileToWrite.chunkSize)
var count: Int
while (inputStream.read(buf).also { count = it } > 0) {
     outputStream.write(buf, 0, count)
}
```

magnusja, 2020-05-13:
Hey there,

so first of all I would make sure it is working properly with Total Commander (USB plugin) and USB Media Explorer. If they are working fine we can be sure that there is something in libaums we can do about it.

## Related issues

There are some related issues like #237 and #209. The situation got better with #242 but is not entirely solved :/

## libusb

Are you using libusb? If not please check it out here https://github.com/magnusja/libaums/tree/develop/libusbcommunication

This definitely helped to significantly reduce the occurrences of this issue.

https://github.com/magnusja/libaums#troubleshooting

smoore6, 2020-05-13:
I tested with Total Commander and USB Media Explorer and it worked fine. I see the issue when I try and copy up to 500 pictures to the drive. I was able to copy ~1500 using Total Commander/USB Media Exploere.

I also tried libusb. There did seem to be a performance improvement, but the issue persists. Here is an updated stack trace:

D/ClusterChain: Init a cluster chain, reading from FAT
D/ClusterChain: Finished init of a cluster chain
D/ClusterChain: grow chain
D/FsInfoStructure: writing to device
E/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):73 libusb_bulk_transfer returned -7
E/ScsiBlockDevice: Error transferring command; errno 0 null
D/ScsiBlockDevice: Reset bulk-only mass storage
W/ScsiBlockDevice: sending bulk only mass storage request
D/ScsiBlockDevice: Trying to clear halt on both endpoints
W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeClearHalt(JNIEnv *, jobject, jlong, jint):98 libusb reset
D/ContentValues: libusb clearFeatureHalt returned kotlin.Unit
W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeClearHalt(JNIEnv *, jobject, jlong, jint):98 libusb reset
D/ContentValues: libusb clearFeatureHalt returned kotlin.Unit
V/InputMethodManager: Starting input: tba=android.view.inputmethod.EditorInfo@4742f9a nm : com.simplifieditproducts.picturekeeperconnect ic=null
I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
D/ConnectivityManager: unregisterNetworkCallback; CallingUid : 10211, CallingPid : 31846
V/InputMethodManager: Starting input: tba=android.view.inputmethod.EditorInfo@efb20ab nm : com.simplifieditproducts.picturekeeperconnect ic=null
I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
D/ConnectivityManager: requestNetwork; CallingUid : 10211, CallingPid : 31846
W/System.err: java.lang.IllegalArgumentException: Expected receiver of type com.android.vending.billing.IInAppBillingService, but got android.os.BinderProxy
W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
W/System.err:     at com.onesignal.TrackGooglePurchase$2.run(TrackGooglePurchase.java:146)
W/System.err:     at java.lang.Thread.run(Thread.java:764)
D/ViewRootImpl@966f4a8[MainActivityBilling]: MSG_RESIZED_REPORT: frame=Rect(0, 0 - 720, 1280) ci=Rect(0, 48 - 0, 0) vi=Rect(0, 48 - 0, 0) or=1
D/ViewRootImpl@966f4a8[MainActivityBilling]: Relayout returned: old=[0,0][720,1280] new=[0,0][720,1280] result=0x3 surface={valid=true 3421589504} changed=false
E/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):73 libusb_bulk_transfer returned -7
E/ScsiBlockDevice: Error transferring command; errno 0 null
D/ScsiBlockDevice: Trying to reset the device
W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeReset(JNIEnv *, jobject, jlong):92 libusb reset

magnusja, 2020-05-25:
Is that the whole stack trace? Does not seem that it crashed up to that point.

With the 64GB drives you provided I get this:
```
2020-05-25 13:30:01.756 459-2890/system_process D/WificondControl: Scan result ready event
2020-05-25 13:31:55.217 3640-3843/com.github.mjdev.usbfileman E/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):73 libusb_bulk_transfer returned -7
2020-05-25 13:31:55.226 3640-3843/com.github.mjdev.usbfileman E/ScsiBlockDevice: Error transferring command; errno 0 null
2020-05-25 13:31:55.226 3640-3843/com.github.mjdev.usbfileman D/ScsiBlockDevice: Reset bulk-only mass storage
2020-05-25 13:31:55.226 3640-3843/com.github.mjdev.usbfileman W/ScsiBlockDevice: sending bulk only mass storage request
2020-05-25 13:31:55.687 3640-3843/com.github.mjdev.usbfileman D/ScsiBlockDevice: Trying to clear halt on both endpoints
2020-05-25 13:31:55.687 3640-3843/com.github.mjdev.usbfileman W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeClearHalt(JNIEnv *, jobject, jlong, jint):98 libusb clear halt
2020-05-25 13:31:55.687 3640-3843/com.github.mjdev.usbfileman D/LibusbCommunication: libusb clearFeatureHalt returned -5
2020-05-25 13:31:55.687 3640-3843/com.github.mjdev.usbfileman W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeClearHalt(JNIEnv *, jobject, jlong, jint):98 libusb clear halt
2020-05-25 13:31:55.687 3640-3843/com.github.mjdev.usbfileman D/LibusbCommunication: libusb clearFeatureHalt returned -5
2020-05-25 13:32:00.019 359-456/? D/hwcomposer: hw_composer sent 6 syncs in 691s
2020-05-25 13:32:01.288 3640-3843/com.github.mjdev.usbfileman E/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):73 libusb_bulk_transfer returned -7
2020-05-25 13:32:01.288 3640-3843/com.github.mjdev.usbfileman E/ScsiBlockDevice: Error transferring command; errno 0 null
2020-05-25 13:32:01.288 3640-3843/com.github.mjdev.usbfileman D/ScsiBlockDevice: Trying to reset the device
2020-05-25 13:32:01.293 3640-3843/com.github.mjdev.usbfileman W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeReset(JNIEnv *, jobject, jlong):92 libusb reset
2020-05-25 13:32:02.831 3640-3843/com.github.mjdev.usbfileman D/LibusbCommunication: libusb reset returned -5
2020-05-25 13:32:02.835 459-667/system_process D/UsbHostManager: Removed device at /dev/bus/usb/001/004: Picture Keeper Connect
2020-05-25 13:32:02.835 459-667/system_process I/UsbAlsaManager: USB Audio Device Removed: null
2020-05-25 13:32:02.836 3640-3843/com.github.mjdev.usbfileman E/MainActivity: error copying!
    java.io.IOException: Could not claim interface, errno: 0 null
        at me.jahnen.libaums.libusbcommunication.LibusbCommunication.resetDevice(LibusbCommunication.kt:91)
        at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:163)
        at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.write(ScsiBlockDevice.kt:282)
        at com.github.mjdev.libaums.driver.ByteBlockDevice.write(ByteBlockDevice.kt:104)
        at com.github.mjdev.libaums.fs.fat32.ClusterChain.write$libaums_debug(ClusterChain.kt:190)
        at com.github.mjdev.libaums.fs.fat32.FatFile.write(FatFile.kt:114)
        at com.github.mjdev.libaums.fs.UsbFileOutputStream.write(UsbFileOutputStream.kt:75)
        at java.io.BufferedOutputStream.flushBuffer(BufferedOutputStream.java:82)
        at java.io.BufferedOutputStream.write(BufferedOutputStream.java:126)
        at com.github.mjdev.libaums.usbfileman.MainActivity$CopyFolderToUsbTask.copyFile(MainActivity.java:564)
        at com.github.mjdev.libaums.usbfileman.MainActivity$CopyFolderToUsbTask.copyDir(MainActivity.java:545)
        at com.github.mjdev.libaums.usbfileman.MainActivity$CopyFolderToUsbTask.copyDir(MainActivity.java:543)
        at com.github.mjdev.libaums.usbfileman.MainActivity$CopyFolderToUsbTask.doInBackground(MainActivity.java:589)
        at com.github.mjdev.libaums.usbfileman.MainActivity$CopyFolderToUsbTask.doInBackground(MainActivity.java:517)
        at android.os.AsyncTask$2.call(AsyncTask.java:333)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
```
It crashes because it cannot reclaim the USB interface, somehow Android rmeoves the device from the system.

The 32GB volume seems to work fine though, can you confirm that?

magnusja, 2020-05-26:
After some more testing I am arriving at the following conclusions:

1. At some point communication is timing out, but no clue why. This can happen immediately or relatively late when copying a larger file (1.5GB in my case), but it *will* happen.
2. The reset and clear endpoint procedure never results in a successful write afterwards (in other words it is useless)
3. The `resetDevice` procedure always results in the need for reenumeration which interrupts file transfer.
4. The TotalCOmmander plugin crashes or hangs as well (no successful transfer of the 1.5GB file) although it usually is able to transfer much more data before crashing than libaums. Their reset and clear endpoint procedure is useless as well. 

### Logs

#### libaums with libusb crashing
1. Error could not claim interface
```
2020-05-26 16:28:34.178 3738-3738/com.github.mjdev.usbfileman I/UsbMassStorageDevice: Found usb interface: UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=8,mSubclass=6,mProtocol=80,mEndpoints=[
    UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0]
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
2020-05-26 16:28:34.179 3738-3738/com.github.mjdev.usbfileman I/UsbMassStorageDevice: Found usb endpoint: UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0]
2020-05-26 16:28:34.179 3738-3738/com.github.mjdev.usbfileman I/UsbMassStorageDevice: Found usb endpoint: UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]
2020-05-26 16:28:34.180 354-3677/system_process I/ActivityManager: START u0 {flg=0x10000000 cmp=com.android.systemui/.usb.UsbPermissionActivity (has extras)} from uid 1000
2020-05-26 16:28:34.182 3738-3738/com.github.mjdev.usbfileman D/MainActivity: USB device attached
2020-05-26 16:28:34.183 3738-3738/com.github.mjdev.usbfileman I/UsbMassStorageDevice: found usb device: /dev/bus/usb/001/004=UsbDevice[mName=/dev/bus/usb/001/004,mVendorId=2316,mProductId=17733,mClass=0,mSubclass=0,mProtocol=0,mManufacturerName=Simplified IT Products,mProductName=Picture Keeper Connect,mVersion=11.00,mSerialNumber=04112402000029,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=150,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=8,mSubclass=6,mProtocol=80,mEndpoints=[
    UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0]
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]]]]
2020-05-26 16:28:34.183 3738-3738/com.github.mjdev.usbfileman I/UsbMassStorageDevice: Found usb interface: UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=8,mSubclass=6,mProtocol=80,mEndpoints=[
    UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0]
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
2020-05-26 16:28:34.183 3738-3738/com.github.mjdev.usbfileman I/UsbMassStorageDevice: Found usb endpoint: UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0]
2020-05-26 16:28:34.183 3738-3738/com.github.mjdev.usbfileman I/UsbMassStorageDevice: Found usb endpoint: UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]
2020-05-26 16:28:34.216 668-668/com.android.systemui W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@e43955c
2020-05-26 16:28:34.259 388-502/? D/SurfaceFlinger: duplicate layer name: changing com.android.systemui/com.android.systemui.usb.UsbPermissionActivity to com.android.systemui/com.android.systemui.usb.UsbPermissionActivity#1
2020-05-26 16:28:34.426 3738-3864/com.github.mjdev.usbfileman E/UsbDeviceConnectionJNI: device is closed in native_claim_interface
2020-05-26 16:28:34.750 668-668/com.android.systemui I/Choreographer: Skipped 30 frames!  The application may be doing too much work on its main thread.
2020-05-26 16:28:34.912 354-509/system_process I/ActivityManager: Displayed com.android.systemui/.usb.UsbPermissionActivity: +724ms
2020-05-26 16:28:34.924 354-509/system_process I/ActivityManager: Displayed com.android.systemui/.usb.UsbResolverActivity: +725ms
2020-05-26 16:28:34.924 354-509/system_process I/ActivityManager: Displayed com.android.systemui/.usb.UsbPermissionActivity: +730ms
2020-05-26 16:28:35.244 3738-3864/com.github.mjdev.usbfileman E/UsbDeviceConnectionJNI: device is closed in native_claim_interface
2020-05-26 16:28:35.245 3738-3864/com.github.mjdev.usbfileman E/MainActivity: error copying!
    java.io.IOException: Could not claim interface, errno: 0 null
        at me.jahnen.libaums.libusbcommunication.LibusbCommunication.resetDevice(LibusbCommunication.kt:93)
        at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:164)
        at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.write(ScsiBlockDevice.kt:283)
        at com.github.mjdev.libaums.driver.ByteBlockDevice.write(ByteBlockDevice.kt:104)
        at com.github.mjdev.libaums.fs.fat32.ClusterChain.write$libaums_debug(ClusterChain.kt:232)
        at com.github.mjdev.libaums.fs.fat32.FatDirectory.write$libaums_debug(FatDirectory.kt:299)
        at com.github.mjdev.libaums.fs.fat32.FatDirectory.createFile(FatDirectory.kt:319)
        at com.github.mjdev.libaums.fs.fat32.FatDirectory.createFile(FatDirectory.kt:36)
        at com.github.mjdev.libaums.usbfileman.MainActivity$CopyFolderToUsbTask.copyFile(MainActivity.java:553)
[... cut ...]
        at de.hechler.tcplugins.usbstick.PluginFunctions.putFile(PluginFunctions.java:874)
        at com.android.tcplugins.FileSystem.IPluginFunctions$Stub.onTransact(IPluginFunctions.java:122)
        at android.os.Binder.execTransact(Binder.java:731)
```

pcap: https://nc.jahnen.me/s/tnEeFNWTkDGgBf8

2. Hanging

pcap: https://nc.jahnen.me/s/cHSAFSZSGXnEFsF

### Questions
* Why can transfer TotalCommander much more data without a the USB device timing out?
* Why is the pkacage length differerent? (shorter and always different with libaums, whereas with TC usually length 64)

haoxi911, 2020-05-27:
Hey @magnusja 

Thanks for your inputs and analysis, it is very helpful. 

Just adding some more background information:

- The issue may also happen on 32GB, but with a much lower reproducible rate.
- The issue doesn't have such a high reproducible rate on our old 64GB drives, but with our new 64GB drives the reproducible rate is very high. We've confirmed with the factory, the only major change they made was replacing the Micron brand NAND flash with a Sandisk brand one.

@smoore6 and I also have a new finding today, that the issue does NOT happen on USB 3.0 phones, we have two phones (Samsung Galaxy S8 and LG G6) which supports USB 3.0 SuperSpeed, and we are NOT able to reproduce the issue on them.

On the other hand, we can easily reproduce the issue on many USB 2.0 phones, and we realized some Android phones with Type-C connectors may still run on USB 2.0 speed. 

The way we tell if the phone is USB 2.0 or 3.0 is using an app called '[USB Device Info](https://play.google.com/store/apps/details?id=aws.apps.usbDeviceEnumerator&hl=en_US)', it will show the 'Max Packet Size' property of the device, which will be 512 for USB 2.0 (High Speed), and 1024 for USB 3.0 (SuperSpeed). 

[image]

Some Questions:

- Could it be power consumption related? We know that different NAND flashes may have different needs on power.
- Why SAF works but libaums doesn't?
- Is 'Max Packet Size' something that makes the difference? Or is it simply because USB 3.0 devices come with a high current than USB 2.0?

magnusja, 2020-06-04:
> @smoore6 and I also have a new finding today, that the issue does NOT happen on USB 3.0 phones, we have two phones (Samsung Galaxy S8 and LG G6) which supports USB 3.0 SuperSpeed, and we are NOT [...]

That is quite interesting, unfortunately I cannot test this. In genymotion/VirtualBox I can only successfully attach a USB 2 host controller.

> Some Questions:
> 
>     * Could it be power consumption related? We know that different NAND flashes may have different needs on power.
I dont think so as this happens on my desktop machine and an emulator as well.
> 
>     * Why SAF works but libaums doesn't?
Well, I guess that SAF is using features of the Linux kernel to mount USB devices and not the Android USB host API.
> 
>     * Is 'Max Packet Size' something that makes the difference? Or is it simply because USB 3.0 devices come with a high current than USB 2.0?
For our problem this should not matter. WIth a larger packet size transfers are usually faster.

haoxi911, 2020-06-04:
Yea, the issue seems to be USB 2.0 only. 

As you mentioned above, the Android system disconnected the USB device while the issue occurs. I am thinking what might cause the disconnection, will it disconnect when we send an invalid SCSI command?

I have a Beagle 480 USB Analyzer and I used it to capture USB packets, the highlighted line in the image below should be a WRITE 10, however its Command Status Wrapper was not there for some reason.

I have very limited knowledge on SCSI and USB, hopefully this won't mislead the discussion, please disregard if it doesn't make any sense :-)

[image]

Gen0, 2020-06-04:
Coming in from the above linked bugreport, I am experiencing the same bug with a USB-C native 32GB pendrive. (SanDisk dual drive). Its rather consistently failing ISO writes between 1gb and 2gb through the transfer. 

Unlike most USB3 drives, this is natively connecting to the phone without a C to female 2.0 adapter and has a USB3 on the other end likely wired to the same points on the PCB. It gets super hot very quick, and my theory is the USB interface on the drive side is overheating. My programming lingo is terrible so sorry in advance. 

Can the transfer be trapped at the failed write then made to wait 10 seconds before searching for the device and attempting to repeat last write and continue? If so, we could eliminate heat as a possible factor, even if we expect it to have to do that a few times to complete a burn process.

If it makes the difference, then we could have a permanent fix in the form or rate limiting the transfers

Depau, 2020-06-04:
> Can the transfer be trapped at the failed write then made to wait 10 seconds before searching for the device and attempting to repeat last write and continue? If so, we could eliminate heat as a possible factor, even if we expect it to have to do that a few times to complete a burn process.

This has been tried, at least by me since many times I stepped around in the debugger while this issue was going on. No, waiting won't do any good.

Can you please attach here the logcat including the part before the issue occurs? I'm expecting to see some attempted recovery steps.

Also if your device is rooted and you can provide the kernel log, that would help too.

Depau, 2020-06-04:
Relevant code, if you wanna give it a look:
https://github.com/magnusja/libaums/blob/38d4a6b6f9e602f6148a5cc2234cd8c3382e86a3/libusbcommunication/src/main/java/me/jahnen/libaums/libusbcommunication/LibusbCommunication.kt#L81-L98

Gen0, 2020-06-05:
Unfortunately this device doesn't have working root yet. 

logcat libusb:I libaum:I eu.depau.etchdroid:I *:S  comes up blank. 🙄

Depau, 2020-06-06:
> Will update when Huawei sort me a BL code.

This may be the reason why you don't get the logs, at my old company I heard stories that you had to get into this weird "factory testing" menu in order to get full logcats on Huaweis. They didn't like Huawei :D

I may want to add to my backlog the implementation of some way to also log somewhere else so users don't have to deal with these shitty phone companies shenanigans 🤦‍♂️ Hopefully now that Google cut them off less people will buy Huawei devices, but since I now own a Xiaomi I know well MIUI sucks just as much. I'm on Lineage right now.

wulileon, 2020-08-26:
Hi  @magnusja
I got the same issue:
first,  it work fine when plug in;

wulileon, 2020-08-26:
Hi @magnusja
I got the same issue:
1st, it work fine when plug in;
2, the problem occurred a few minutes after the refresh was stopped 

### I use  Kingston 500GB SSD with nvme bridge. 

 I/KMarsLib: (HomeFragment.kt:399) ==> list path: /
 I/KMarsLib: (HomeFragment.kt:399) ==> list path: /backup
 I/KMarsLib: (HomeFragment.kt:399) ==> list path: /backup/2020-08-26
 I/KMarsLib: (HomeFragment.kt:399) ==> list path: /backup/2020-08-26/picture
 W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@619bfb5
 W/Choreographer: Already have a pending vsync event.  There should only be one at a time.
 E/ScsiBlockDevice: Error transferring command; errno 0 null
 W/ScsiBlockDevice: sending bulk only mass storage request
 I/KMarsLib: (HomeFragment.kt:399) ==> list path: /
 E/ScsiBlockDevice: Error transferring command; errno 0 null
 E/ScsiBlockDevice: Error transferring command; errno 0 null
 W/System.err: java.io.IOException: Could not write to device, result == -1 errno 0 null
 W/System.err:     at com.github.mjdev.libaums.usb.JellyBeanMr2Communication.bulkOutTransfer(JellyBeanMr2Communication.kt:37)
 W/System.err:     at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferOneCommand(ScsiBlockDevice.kt:196)
 W/System.err:     at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:143)
 W/System.err:     at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.read(ScsiBlockDevice.kt:265)
 W/System.err:     at com.github.mjdev.libaums.driver.ByteBlockDevice.read(ByteBlockDevice.kt:54)
 W/System.err:     at com.github.mjdev.libaums.fs.fat32.ClusterChain.read$libaums_release(ClusterChain.kt:154)
 W/System.err:     at com.github.mjdev.libaums.fs.fat32.FatDirectory.readEntries(FatDirectory.kt:161)
 W/System.err:     at com.github.mjdev.libaums.fs.fat32.FatDirectory.init(FatDirectory.kt:144)
 W/System.err:     at com.github.mjdev.libaums.fs.fat32.FatDirectory.listFiles(FatDirectory.kt:405)
 W/System.err:     at com.storage.filesrequest.action.usb.UsbActionRecursiveListFiles.doListAction(UsbActionRecursiveListFiles.kt:55)
 W/System.err:     at com.storage.filesrequest.action.usb.UsbActionRecursiveListFiles.doListAction(UsbActionRecursiveListFiles.kt:35)
 W/System.err:     at com.storage.filesrequest.action.usb.UsbActionRecursiveListFiles.doAction(UsbActionRecursiveListFiles.kt:14)
 W/System.err:     at com.storage.filesrequest.usb.libaums.AumsDiskDelegateImpl.recursiveList(AumsDiskDelegateImpl.kt:89)
 W/System.err:     at com.storage.filesrequest.usb.FRUsbManager.recursiveList(FRUsbManager.kt:238)
 W/System.err:     at com.storage.nvbackup.ui.mainui.PhotoFragment.listType(PhotoFragment.kt:250)
 W/System.err:     at com.storage.nvbackup.ui.mainui.PhotoFragment.access$listType(PhotoFragment.kt:44)
 W/System.err:     at com.storage.nvbackup.ui.mainui.PhotoFragment$doListAsync$1.subscribe(PhotoFragment.kt:223)
 W/System.err:     at io.reactivex.internal.operators.observable.ObservableCreate.subscribeActual(ObservableCreate.java:40)
 W/System.err:     at io.reactivex.Observable.subscribe(Observable.java:12030)
 W/System.err:     at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeTask.run(ObservableSubscribeOn.java:96)
 W/System.err:     at io.reactivex.Scheduler$DisposeTask.run(Scheduler.java:579)
 W/System.err:     at io.reactivex.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:66)
 W/System.err:     at io.reactivex.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:57)
 W/System.err:     at java.util.concurrent.FutureTask.run(FutureTask.java:266)
 W/System.err:     at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
 W/System.err:     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
 W/System.err:     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
 W/System.err:     at java.lang.Thread.run(Thread.java:919)

### when use libusb:

I/KMarsLib: (HomeFragment.kt:399) ==> list path: /
I/KMarsLib: (HomeFragment.kt:399) ==> list path: /backup
E/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):73 libusb_bulk_transfer returned -7
E/ScsiBlockDevice: Error transferring command; errno 0 null
W/ScsiBlockDevice: sending bulk only mass storage request
W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeClearHalt(JNIEnv *, jobject, jlong, jint):98 libusb reset
W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeClearHalt(JNIEnv *, jobject, jlong, jint):98 libusb reset
E/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):73 libusb_bulk_transfer returned -7
E/ScsiBlockDevice: Error transferring command; errno 0 null
W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeReset(JNIEnv *, jobject, jlong):92 libusb reset
E/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):73 libusb_bulk_transfer returned -7
E/ScsiBlockDevice: Error transferring command; errno 0 null
[... cut ...]
                }
            } else {
                var p = path
                if (path.startsWith("/")) {
                    p = path.substring(1)
                }
                val usbFile = root.search(p)
                usbFile?.let {file ->

                    for (f in file.listFiles()) {
                        val e = FRFile.parse(it.volumeLabel, path, f)
                        list.add(e)
                    }

                }
            }
        }

        return list
    }