magnusja/libaums #254: error setting up device

happyAshley, 2020-05-15:
when i delete all files and dirs on usb, it crashed.  And i have to format the usb again：
the log as below:
2020-05-13 09:00:37.842 9754-9754/com.github.mjdev.usbfileman E/MainActivity: error setting up device
java.io.IOException: Could not read from device, result == -1 errno 0 null
    at com.github.mjdev.libaums.usb.JellyBeanMr2Communication.bulkInTransfer(JellyBeanMr2Communication.kt:47)
    at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:161)
    at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.read(ScsiBlockDevice.kt:248)
    at com.github.mjdev.libaums.driver.ByteBlockDevice.read(ByteBlockDevice.kt:82)
    at com.github.mjdev.libaums.fs.fat32.Fat32FileSystem$Companion.read(Fat32FileSystem.kt:117)
    at com.github.mjdev.libaums.fs.fat32.Fat32FileSystemCreator.read(Fat32FileSystemCreator.kt:21)
    at com.github.mjdev.libaums.fs.FileSystemFactory.createFileSystem(FileSystemFactory.kt:57)
    at com.github.mjdev.libaums.partition.fs.FileSystemPartitionTableCreator.read(FileSystemPartitionTableCreator.kt:21)
    at com.github.mjdev.libaums.partition.PartitionTableFactory.createPartitionTable(PartitionTableFactory.kt:62)
    at com.github.mjdev.libaums.UsbMassStorageDevice.setupDevice(UsbMassStorageDevice.kt:157)
    at com.github.mjdev.libaums.UsbMassStorageDevice.init(UsbMassStorageDevice.kt:100)
    at com.github.mjdev.libaums.usbfileman.MainActivity.setupDevice(MainActivity.java:761)
    at com.github.mjdev.libaums.usbfileman.MainActivity.access$000(MainActivity.java:99)
    at com.github.mjdev.libaums.usbfileman.MainActivity$1.onReceive(MainActivity.java:128)
    at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0(LoadedApk.java:1391)
    at android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run(Unknown Source:2)
    at android.os.Handler.handleCallback(Handler.java:873)
    at android.os.Handler.dispatchMessage(Handler.java:99)
    at android.os.Looper.loop(Looper.java:193)
    at android.app.ActivityThread.main(ActivityThread.java:6718)
    at java.lang.reflect.Method.invoke(Native Method)
    at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:491)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)

happyAshley, 2020-05-15:
usb device  works well on Windows platform。

magnusja, 2020-05-18:
Please see if https://github.com/magnusja/libaums/tree/develop/libusbcommunication fixes that probem

happyAshley, 2020-05-19:
no，libusb can not fix this problem。