magnusja/libaums #255: Deleting multiple files is slow

petrs143, 2020-05-24:
Hello, when I delete 2500 files in one folder, on some devices, it takes more than 30 minutes. In my opinion it is caused by calling `write()` on each `delete()`. I' ve already tried to call only `removeEntry()` on each file and then call `write()`, but it corrupts my drive somehow and deletes more files than should delete. Is there any solution?

magnusja, 2020-05-24:
In theory this should be possible. Can you show some code?

petrs143, 2020-05-25:
In `FatDirectory`:

``` kotlin
  fun deleteFiles(files: List<FatFile>) {
        files.forEach {
            it.initChain()
            removeEntry(it.entry)
        }
        write()
        files.forEach {
            it.chain.length = 0
        }
    }
```

Usage:
``` kotlin
val files = folder.listFiles().toList()
folder.deleteFiles(files.filter { /* Some filtration by name */ })
```

magnusja, 2020-05-25:
Does it work if you do not set the chain to `0`?

petrs143, 2020-05-25:
No, it still deletes more files than it should.

petrs143, 2020-06-12:
@magnusja I've noticed that `FatFile.delete()` doesn't work for me too. But only for 400+ files in a folder.

# My usage

- Create a folder with 400 files on a PC
- In my app, I list the folder and delete 200 files (like the previous usage)

But when I open USB drive on a PC whole folder disappear and remaining files are in LOST.DIR folder.