magnusja/libaums #258: use libusb on no rooted android。libusb_init returned -1.permission deny

happyAshley, 2020-05-29:
2020-05-29 09:26:18.401 2012-2012/com.github.mjdev.usbfileman W/jdev.usbfileman: type=1400 audit(0.0:1079): avc: denied { read } for name="usb" dev="tmpfs" ino=10466 scontext=u:r:untrusted_app:s0:c119,c256,c512,c768 tcontext=u:object_r:usb_device:s0 tclass=dir permissive=0
2020-05-29 09:26:18.408 2012-2012/com.github.mjdev.usbfileman E/native_libusbcom: jboolean Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeInit(JNIEnv *, jobject, jint, jlongArray):14 libusb_init returned -1
2020-05-29 09:26:18.410 2012-2012/com.github.mjdev.usbfileman E/MainActivity: error setting up device
    java.io.IOException: libusb init failed
        at me.jahnen.libaums.libusbcommunication.LibusbCommunication.<init>(LibusbCommunication.kt:34)
        at me.jahnen.libaums.libusbcommunication.LibusbCommunicationCreator.create(LibusbCommunication.kt:113)
        at com.github.mjdev.libaums.usb.UsbCommunicationFactory.createUsbCommunication(UsbCommunicationFactory.kt:59)
        at com.github.mjdev.libaums.UsbMassStorageDevice.setupDevice(UsbMassStorageDevice.kt:126)
        at com.github.mjdev.libaums.UsbMassStorageDevice.init(UsbMassStorageDevice.kt:100)
        at com.github.mjdev.libaums.usbfileman.MainActivity.setupDevice(MainActivity.java:767)
        at com.github.mjdev.libaums.usbfileman.MainActivity.access$000(MainActivity.java:103)
        at com.github.mjdev.libaums.usbfileman.MainActivity$1.onReceive(MainActivity.java:134)
        at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0(LoadedApk.java:1391)
        at android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run(Unknown Source:2)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6718)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:491)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)

Depau, 2020-07-29:
Can you add some steps for reproducing the issue? A code snippet?

Depau, 2020-07-29:
I saw you also reported the same issue for the Android API communication implementation, and it's exactly the same error.
Have you tried it on another device?

Please try the following:

- Try with another USB drive on the same phone
- Try with another app, another USB device on the same phone (i.e. a keyboard or mouse)
- Try the same USB drive on a different phone