magnusja/libaums #263: java.util.ConcurrentModificationException

happyAshley, 2020-08-02:
### Problem
An error occurred while uploading the file to usb device.

### Expected behavior

### Actual behavior

#### Stacktrace of Excpetion

```
(if exists)
```

### Code where problem occurs
2020-08-03 03:10:54.183 12045-12082/com.dada D/FatDirectory: adding entry: [FatLfnDirectoryEntry getName()=2202.zip] with short name: 220000~0.ZIP
2020-08-03 03:10:54.184 12045-12081/com.dada W/System.err: java.util.ConcurrentModificationException
2020-08-03 03:10:54.185 12045-12081/com.dada W/System.err:     at java.util.ArrayList$Itr.next(ArrayList.java:860)
2020-08-03 03:10:54.185 12045-12081/com.dada W/System.err:     at com.github.mjdev.libaums.fs.fat32.FatDirectory.write$libaums_release(FatDirectory.kt:289)
2020-08-03 03:10:54.185 12045-12081/com.dada W/System.err:     at com.github.mjdev.libaums.fs.fat32.FatDirectory.createFile(FatDirectory.kt:319)
2020-08-03 03:10:54.185 12045-12081/com.dada W/System.err:     at com.github.mjdev.libaums.fs.fat32.FatDirectory.createFile(FatDirectory.kt:36)
2020-08-03 03:10:54.185 12045-12081/com.dada W/System.err:     at com.dada.UsbUtils.UsbFileUtils.saveSDFileToUsb(UsbFileUtils.java:43)
2020-08-03 03:10:54.186 12045-12081/com.dada W/System.err:     at com.dada.UsbUtils.UsbActivity$1.run(UsbActivity.java:94)
2020-08-03 03:10:54.186 12045-12081/com.dada W/System.err:     at java.lang.Thread.run(Thread.java:764)

magnusja, 2020-08-02:
Can you show some code? Seems like you are using multiple threads, which is not allowed. Try to stick to just one thread using libaums

happyAshley, 2020-08-03:
yes, I just used multithreading to transfer files。

magnusja, 2020-08-03:
Only use one thread for calling into libaums.

closed 2020-08-03 by magnusja