magnusja/libaums #266: usbfileman can't copy file to exFat fs usb drive

wulileon, 2020-08-26:
### Problem
usbfileman can read file & folder in this usb, but when I try to "copy To USB"  , usbfileman crash
usb drive:  Toshiba 32GB usb3.0
android:  Android 10

BTW,  it takes a few seconds to show file list,  ANR is prone to occur

#### Stacktrace of Excpetion

2020-08-26 14:19:06.330 10227-10227/com.github.mjdev.usbfileman E/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):73 libusb_bulk_transfer returned -7
2020-08-26 14:19:06.341 10227-10227/com.github.mjdev.usbfileman E/ScsiBlockDevice: Error transferring command; errno 0 null
2020-08-26 14:19:06.342 10227-10227/com.github.mjdev.usbfileman W/ScsiBlockDevice: sending bulk only mass storage request
2020-08-26 14:19:09.744 10227-10227/com.github.mjdev.usbfileman W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeClearHalt(JNIEnv *, jobject, jlong, jint):98 libusb reset
2020-08-26 14:19:09.746 10227-10227/com.github.mjdev.usbfileman W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeClearHalt(JNIEnv *, jobject, jlong, jint):98 libusb reset
2020-08-26 14:19:11.427 10227-10238/com.github.mjdev.usbfileman I/jdev.usbfilema: Thread[5,tid=10238,WaitingInMainSignalCatcherLoop,Thread*=0x77ec23c800,peer=0x12d83468,"Signal Catcher"]: reacting to signal 3
2020-08-26 14:19:11.527 10227-10238/com.github.mjdev.usbfileman I/jdev.usbfilema: Wrote stack traces to tombstoned
2020-08-26 14:19:15.253 10227-10227/com.github.mjdev.usbfileman E/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):73 libusb_bulk_transfer returned -7
2020-08-26 14:19:15.254 10227-10227/com.github.mjdev.usbfileman E/ScsiBlockDevice: Error transferring command; errno 0 null
2020-08-26 14:19:15.261 10227-10227/com.github.mjdev.usbfileman W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeReset(JNIEnv *, jobject, jlong):92 libusb reset
2020-08-26 14:19:21.602 10227-10227/com.github.mjdev.usbfileman E/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):73 libusb_bulk_transfer returned -7
2020-08-26 14:19:21.604 10227-10227/com.github.mjdev.usbfileman E/ScsiBlockDevice: Error transferring command; errno 0 null
2020-08-26 14:19:21.619 10227-10227/com.github.mjdev.usbfileman E/MainActivity: error initializing adapter!
    java.io.IOException: libusb returned -7 in bulk out transfer
        at me.jahnen.libaums.libusbcommunication.LibusbCommunication.bulkOutTransfer(LibusbCommunication.kt:59)
        at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferOneCommand(ScsiBlockDevice.kt:196)
        at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:143)
        at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.read(ScsiBlockDevice.kt:265)
        at com.github.mjdev.libaums.driver.ByteBlockDevice.read(ByteBlockDevice.kt:54)
        at com.github.magnusja.libaums.javafs.wrapper.device.FSBlockDeviceWrapper.read(FSBlockDeviceWrapper.java:59)
        at org.jnode.fs.exfat.DeviceAccess.read(DeviceAccess.java:102)
        at org.jnode.fs.exfat.ExFatSuperBlock.readCluster(ExFatSuperBlock.java:149)
        at org.jnode.fs.exfat.DirectoryParser.init(DirectoryParser.java:94)
        at org.jnode.fs.exfat.DirectoryParser.create(DirectoryParser.java:61)
        at org.jnode.fs.exfat.NodeDirectory.<init>(NodeDirectory.java:60)
        at org.jnode.fs.exfat.NodeDirectory.<init>(NodeDirectory.java:46)
        at org.jnode.fs.exfat.NodeEntry.getDirectory(NodeEntry.java:118)
        at com.github.magnusja.libaums.javafs.wrapper.fs.UsbFileWrapper.<init>(UsbFileWrapper.java:36)
        at com.github.magnusja.libaums.javafs.wrapper.fs.UsbFileWrapper.listFiles(UsbFileWrapper.java:136)
        at com.github.mjdev.libaums.usbfileman.UsbFileListAdapter.refresh(UsbFileListAdapter.java:98)
        at com.github.mjdev.libaums.usbfileman.UsbFileListAdapter.<init>(UsbFileListAdapter.java:87)
        at com.github.mjdev.libaums.usbfileman.MainActivity.onBackPressed(MainActivity.java:1134)
        at android.app.Activity.onKeyUp(Activity.java:3657)
        at android.view.KeyEvent.dispatch(KeyEvent.java:2874)
        at android.app.Activity.dispatchKeyEvent(Activity.java:3974)
        at androidx.core.app.ComponentActivity.superDispatchKeyEvent(ComponentActivity.java:122)
        at androidx.core.view.KeyEventDispatcher.dispatchKeyEvent(KeyEventDispatcher.java:84)
        at androidx.core.app.ComponentActivity.dispatchKeyEvent(ComponentActivity.java:140)
        at androidx.appcompat.app.AppCompatActivity.dispatchKeyEvent(AppCompatActivity.java:535)
        at androidx.appcompat.view.WindowCallbackWrapper.dispatchKeyEvent(WindowCallbackWrapper.java:59)
        at androidx.appcompat.app.AppCompatDelegateImpl$AppCompatWindowCallback.dispatchKeyEvent(AppCompatDelegateImpl.java:2533)
        at com.android.internal.policy.DecorView.dispatchKeyEvent(DecorView.java:375)
        at android.view.ViewRootImpl$ViewPostImeInputStage.processKeyEvent(ViewRootImpl.java:5625)
        at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:5489)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4981)
        at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:5034)
        at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:5000)
        at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:5140)
        at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:5008)
        at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:5197)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4981)
        at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:5034)
        at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:5000)
        at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:5008)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4981)
        at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:5034)
        at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:5000)
        at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:5173)
        at android.view.ViewRootImpl$ImeInputStage.onFinishedInputEvent(ViewRootImpl.java:5334)
        at android.view.inputmethod.InputMethodManager$PendingEvent.run(InputMethodManager.java:3125)
        at android.view.inputmethod.InputMethodManager.invokeFinishedInputEventCallback(InputMethodManager.java:2668)

magnusja, 2020-08-30:
Can you try latest develop branch? there have been some fixes, not sure if they will fix that problem though.

MestevInge, 2020-09-16:
Hello,

Do you have any solution or suggestion actually for writting with extFat system ?
How do work Paragon ? We can mount/unmount partition and R/W easily with the File classe

magnusja, 2020-09-16:
> Hello,
> 
> Do you have any solution or suggestion actually for writting with extFat system ?
> How do work Paragon ? We can mount/unmount partition and R/W easily with the File classe

see here https://github.com/magnusja/libaums/issues/212#issuecomment-623694148

happy to take contributions for that

closed 2023-03-11 by magnusja