magnusja/libaums #268: Coudn't open video file using AsyncHttpServer

spiritedRunning, 2020-09-01:
### Problem
I use the following code to init and start HTTP server as demo does.
`serverService.startServer(file, new AsyncHttpServer(8000));`

But log tells me it hasn't a hostname. So I give a default hostname with “localhost”. But it still not works, however.  The error log below:
`2020-09-01 17:26:00.306 4962-4962/com.github.mjdev.usbfileman D/MainActivity: starting HTTP server
2020-09-01 17:26:00.381 4962-4962/com.github.mjdev.usbfileman I/MainActivity: base url=http://localhost:8000/, name = SUZHOU.MP4
2020-09-01 17:26:00.798 4962-4985/com.github.mjdev.usbfileman D/OpenGLRenderer: endAllActiveAnimators on 0x7f5a764000 (MenuPopupWindow$MenuDropDownListView) with handle 0x7f5b95b480
2020-09-01 17:26:03.614 4962-4962/com.github.mjdev.usbfileman W/IInputConnectionWrapper: reportFullscreenMode on inexistent InputConnection
2020-09-01 17:26:03.665 4962-5216/com.github.mjdev.usbfileman D/AsyncHttpServer: Uri: /SUZHOU.MP4
2020-09-01 17:26:03.665 4962-5216/com.github.mjdev.usbfileman D/UsbFileHttpServer: Searching file on USB (URI: /SUZHOU.MP4)
2020-09-01 17:26:03.665 4962-5216/com.github.mjdev.usbfileman D/UsbFileHttpServer: Serving root file
2020-09-01 17:26:03.684 4962-5216/com.github.mjdev.usbfileman D/ClusterChain: Init a cluster chain, reading from FAT
2020-09-01 17:26:03.685 4962-5216/com.github.mjdev.usbfileman E/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):73 libusb_bulk_transfer returned -1
2020-09-01 17:26:03.688 4962-5216/com.github.mjdev.usbfileman E/ScsiBlockDevice: Error transferring command; errno 0 null
2020-09-01 17:26:03.689 4962-5216/com.github.mjdev.usbfileman D/ScsiBlockDevice: Reset bulk-only mass storage
2020-09-01 17:26:03.689 4962-5216/com.github.mjdev.usbfileman W/ScsiBlockDevice: sending bulk only mass storage request
2020-09-01 17:26:03.701 4962-5216/com.github.mjdev.usbfileman E/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):73 libusb_bulk_transfer returned -1
2020-09-01 17:26:03.701 4962-5216/com.github.mjdev.usbfileman E/ScsiBlockDevice: Error transferring command; errno 0 null
2020-09-01 17:26:03.701 4962-5216/com.github.mjdev.usbfileman D/ScsiBlockDevice: Reset bulk-only mass storage
2020-09-01 17:26:03.701 4962-5216/com.github.mjdev.usbfileman W/ScsiBlockDevice: sending bulk only mass storage request
2020-09-01 17:26:03.714 4962-5216/com.github.mjdev.usbfileman E/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):73 libusb_bulk_transfer returned -1
2020-09-01 17:26:03.714 4962-5216/com.github.mjdev.usbfileman E/ScsiBlockDevice: Error transferring command; errno 0 null
2020-09-01 17:26:03.714 4962-5216/com.github.mjdev.usbfileman D/ScsiBlockDevice: Reset bulk-only mass storage
2020-09-01 17:26:03.714 4962-5216/com.github.mjdev.usbfileman W/ScsiBlockDevice: sending bulk only mass storage request`

I also following the instruction in README,  Init HTTP server in a another way.

`UsbFile file = ... // can be directory or file

HttpServer server = AsyncHttpServer(8000); // port 8000
UsbFileHttpServer fileServer = new UsbFileHttpServer(file, server);
fileServer.start();`
However, It bring me a crash and has error below:
`2020-09-01 17:07:54.376 5571-5571/com.github.mjdev.usbfileman E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.github.mjdev.usbfileman, PID: 5571
    java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String com.github.mjdev.libaums.server.http.UsbFileHttpServer.getBaseUrl()' on a null object reference
        at com.github.mjdev.libaums.usbfileman.MainActivity.startHttpServer(MainActivity.java:1013)
        at com.github.mjdev.libaums.usbfileman.MainActivity.onContextItemSelected(MainActivity.java:926)
        at android.app.Activity.onMenuItemSelected(Activity.java:3268)
        at androidx.fragment.app.FragmentActivity.onMenuItemSelected(FragmentActivity.java:426)
        at androidx.appcompat.app.AppCompatActivity.onMenuItemSelected(AppCompatActivity.java:196)
        at androidx.appcompat.view.WindowCallbackWrapper.onMenuItemSelected(WindowCallbackWrapper.java:109)
        at com.android.internal.policy.PhoneWindow$PhoneWindowMenuCallback.onMenuItemSelected(PhoneWindow.java:3680)
        at com.android.internal.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:761)
        at com.android.internal.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:152)
        at com.android.internal.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:904)
        at com.android.internal.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:894)
        at com.android.internal.view.menu.MenuPopup.onItemClick(MenuPopup.java:128)
        at android.widget.AdapterView.performItemClick(AdapterView.java:310)
        at android.widget.AbsListView.performItemClick(AbsListView.java:1164)
        at android.widget.AbsListView$PerformClick.run(AbsListView.java:3139)
        at android.widget.AbsListView$3.run(AbsListView.java:4054)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6121)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
`

magnusja, 2020-09-10:
Can you try latest develop branch? We fixed some bugs there, but this looks a lot like what we have been discussing here https://github.com/magnusja/libaums/issues

magnusja, 2020-09-10:
Ok I released a new version please try that one

closed 2022-01-22 by magnusja