magnusja/libaums #271: Copy slow

jackyzhougithub, 2020-09-21:
The speed of copying files from a 64GB SanDisk card to the phone is very slow. Is the copy speed related to the size of the SD card? (16G SD card is faster, fs: exfat)

Depau, 2020-09-21:
Hi, can you do an equivalent test using a computer or a device with an internal SD card slot?

Or can you check if the larger one is also slower with EtchDroid, to cross out any issues with the file system implementation since EtchDroid bypasses it? (see my profile or search on google play, keep in mind that it will wipe all the data on the cards and you will need to format them).

While this isn't often true for SSDs, higher capacity SD cards are more likely to make use of MLC/TLC NANDs. They provide double/triple capacity in the same physical volume at the cost of much lower speeds, reliability and lifetime. Therefore it's unlikely that this is a problem with libaums.

jackyzhougithub, 2020-09-21:
Thank you very much for your answer. The copy speed of the slow SD card using ES File Explorer is very fast (80M/s), and the copy speed of libaums is only about 500KB/s. I am very confused about this.

MestevInge, 2020-10-02:
It is because libaums get chunksize of the storage.
(Example : With USB 16G, Libaums have 16K byte for chunksize)
ES File Explorer take another size for copy byte to byte.

Depau, 2020-10-09:
I see.

I also saw someone commented about the block size being an issue and while the comment has been deleted, that might be something interesting to look at.

Are you using the UsbFileMan demo app?

closed 2020-12-14 by jackyzhougithub

devintrue, 2024-08-08:
the same problem. May I ask how it was resolved