magnusja/libaums #274: sample usage request.   (not an issue)

vittalk, 2020-10-11:
### Problem
debug|x86 :-- Detecting CXX compile features - done
debug|x86 :Using libusb path null
debug|x86 :-- Configuring done
debug|x86 :CMake Error at CMakeLists.txt:11 (add_library):
debug|x86 :  Cannot find source file:
debug|x86 :    null/libusb/core.c
debug|x86 :  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
debug|x86 :  .hxx .in .txx
debug|x86 :CMake Error: CMake can not determine linker language for target: libusb
debug|x86 :-- Generating done

### Expected behavior

### Actual behavior

#### Stacktrace of Excpetion

```
(if exists)
```

### Code where problem occurs

Depau, 2020-10-11:
Hi,
Documentation for the libusb module is here: https://github.com/magnusja/libaums/tree/develop/libusbcommunication

If you build the whole project you need to download libusb and add it to your local.properties.

https://github.com/magnusja/libaums/blob/develop/.travis.yml#L24-L26

closed 2020-10-11 by Depau

vittalk, 2020-10-19:
HI Davide,
Thank you. Your suggestion helped me to compile the code. But now I have a
different problem. It fails to install,
I get the conflicting provider error.. I have tried various suggestions
from the internet.. none of them have worked..
Greatly appreciate any help.

Installation did not succeed.
The application could not be installed: INSTALL_FAILED_CONFLICTING_PROVIDER

List of apks:
[0]
'/Users/vittal/AndroidStudioProjects/LIBAUMS_work/libaums-master/app/build/outputs/apk/debug/app-debug.apk'
Installation failed due to: 'null'

Thanks and Best Regards,
-Vittal

vittalk, 2020-10-19:
To continue,
i can send whatever files you want to inspect , or the whole folder (24MB)
if you want to load it on android studio and see ..

Thanks and Best Regards,
-Vittal

magnusja, 2020-10-19:
Do you have another app using libaums on that device? 

Try deleting that part https://github.com/magnusja/libaums/blob/develop/storageprovider/src/main/AndroidManifest.xml#L10-L20

vittalk, 2020-10-19:
Gentlemen,
Thank  you very much. I was able to compile and load the libaums. Greatly
appreciated

Thanks and Best Regards,
-Vittal

vittalk, 2020-10-22:
Hi Davide, Magnus,
Thank for your help. I am able to load the USB drive and display the
directory contents.... I have another question to ask.

I need to read a JSON file stored in the USB flash drive.  I am using
Google's GSON library to read the json file and analyze the content. Below
is the code.
As you can see, 'reader' is used as a parameter to gson.fromJson(). and
reader holds the content of the file extData.json.  I've been trying to
make this code work with
your  UsbFileInputStream but I am unable to do so. Any suggestions please?
Greatly appreciate your help

try {
    // create Gson instance
    val gson = Gson()

    // create a reader

     Reader reader = Files.newBufferedReader(Paths.get("extData.json"));
    // ------------convert JSON file to map ----------
    val map: Map<*, *> = gson.fromJson(reader, MutableMap::class.java)

    val SID_tag : String = "SID"

    // print map entries
    for ((key, value) in map) {
        println(key.toString() + "  =   " + value)
        val tempString : String  = key.toString() + " = " + value
        Log.i("json file content:   ", tempString)
        if (key.toString() == SID_tag) {
            ssID.text = "SID: " + value.toString()
            Log.i("Found SID IS:----  ", value.toString())
        }

    }

    // close reader
    reader.close()
} catch (ex: Exception) {
    ex.printStackTrace()
}

Thanks and Best Regards,
-Vittal

vittalk, 2020-10-22:
Hi Gentlemen,
To summarize my request, I need to read and write a JSON file present in
the USB stick. I might later need to copy and transfer files(for which you
have sample code already available). I am using GSON now, I am open to use
any other library if you can recommend..

Thanks and Best Regards,
-Vittal

magnusja, 2020-10-22:
This should help you out https://github.com/google/gson/issues/187

vittalk, 2020-10-22:
> This should help you out google/gson#187

Thank you so very much. This worked..
Greatly appreciated

Thanks and Best Regards,
-Vittal