magnusja/libaums #275: Error transferring command; errno 0 null

anneyking, 2020-10-13:
I have a removable hard drive, which has two partitions, a public partition and an encrypted partition (unlocked with fingerprints). When the hard drive is plugged into the android phone, the call to API,device.init (), in libaums will always be unsuccessful. Is there any way to ignore encrypted partitions in init () and only init () public partitions?

Log when device.init () is called :

10-13 10:10:02.608 30493-30493/com.example.fingerdisk D/FatDirectory: volume label: NORMAL     
10-13 10:10:02.611 30493-30493/com.example.fingerdisk D/Fat32FileSystem: Fat32BootSector{bytesPerSector=512, sectorsPerCluster=16, reservedSectors=5328, fatCount=2, totalNumberOfSectors=28130112, sectorsPerFat=13720, rootDirStartCluster=2, fsInfoStartSector=1, fatMirrored=true, validFat=0, volumeLabel=''}
10-13 10:10:02.618 30493-30493/com.example.fingerdisk D/ScsiBlockDevice: inquiry response: ScsiInquiryResponse [peripheralQualifier=0, peripheralDeviceType=0, removableMedia=false, spcVersion=6, responseDataFormat=2]
10-13 10:10:02.621 30493-30493/com.example.fingerdisk W/linker: liberrno-lib.so: unused DT entry: type 0x6ffffef5 arg 0x4a0
10-13 10:10:02.621 30493-30493/com.example.fingerdisk W/linker: liberrno-lib.so: unused DT entry: type 0x6ffffffe arg 0x5bc
10-13 10:10:02.621 30493-30493/com.example.fingerdisk W/linker: liberrno-lib.so: unused DT entry: type 0x6fffffff arg 0x2
10-13 10:10:02.621 30493-30493/com.example.fingerdisk E/ScsiBlockDevice: Error transferring command; errno 0 null
10-13 10:10:02.621 30493-30493/com.example.fingerdisk D/ScsiBlockDevice: Reset bulk-only mass storage
10-13 10:10:02.621 30493-30493/com.example.fingerdisk W/ScsiBlockDevice: sending bulk only mass storage request
10-13 10:10:02.621 30493-30493/com.example.fingerdisk D/ScsiBlockDevice: Trying to clear halt on both endpoints
10-13 10:10:02.622 30493-30493/com.example.fingerdisk W/AndroidUsbCommunication: Clearing halt on endpoint UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0] (direction 128)
10-13 10:10:02.622 30493-30493/com.example.fingerdisk W/AndroidUsbCommunication: Clearing halt on endpoint UsbEndpoint[mAddress=2,mAttributes=2,mMaxPacketSize=512,mInterval=0] (direction 0)
10-13 10:10:02.622 30493-30493/com.example.fingerdisk E/ScsiBlockDevice: Error transferring command; errno 0 null
10-13 10:10:02.923 30493-30493/com.example.fingerdisk D/ScsiBlockDevice: Trying to reset the device
10-13 10:10:02.923 30493-30493/com.example.fingerdisk D/AndroidUsbCommunication: Performing native reset
10-13 10:10:03.361 30493-30493/com.example.fingerdisk E/ScsiBlockDevice: Error transferring command; errno 0 null

magnusja, 2020-10-13:
It looks like it is not trying to read the encrypted partition but has problems with the regular one.

Can you try the `libusbcommunication` module?

anneyking, 2020-10-26:
> It looks like it is not trying to read the encrypted partition but has problems with the regular one.
> 
> Can you try the `libusbcommunication` module?

How to do some things with libusbcommunication module, such as getRootDirectory ,listFiles, file.getName?