magnusja/libaums #283: Unit attention (ASC: 40, ASCQ: 0)

xmutzlq, 2021-01-29:
### Problem
UsbMassStorageDevice init, throws Unit attention (ASC: 40, ASCQ: 0)

### Expected behavior

### Actual behavior
when U Disk inserted，it is normal work；then pull plugs U Disk and insert U Disk again，finally  throws Unit attention (ASC: 40, ASCQ: 0)

#### Stacktrace of Excpetion

```
(if exists)
```

### Code where problem occurs
private void setupDevice() {
        LogUtils.debugInfo("zlq", "currentDevice" + currentDevice + "->setupDevice");
        try {
            if(currentDevice == -1) return;
            storageDevices[currentDevice].init();
            LogUtils.debugInfo("zlq", "usb device init success");
            if(storageDevices[currentDevice].getPartitions() != null
                    && storageDevices[currentDevice].getPartitions().size() > 0) {
                currentFs = storageDevices[currentDevice].getPartitions().get(0).getFileSystem();
                if(currentFs != null) {
                    usbRoot = currentFs.getRootDirectory();
                    LogUtils.debugInfo("zlq", "Capacity: " + currentFs.getCapacity());
                    LogUtils.debugInfo("zlq", "Occupied Space: " + currentFs.getOccupiedSpace());
                    LogUtils.debugInfo("zlq", "Free Space: " + currentFs.getFreeSpace());
                    LogUtils.debugInfo("zlq", "Chunk size: " + currentFs.getChunkSize());
                }
            }
        } catch (IOException e) {
            LogUtils.debugInfo("zlq", "setupDeviceError = " + e.getMessage());
            discoverDevice();
        }
    }

Derpalus, 2021-02-10:
I have seen this issue before on some types of SD card readers. The corresponding message is `Not Ready to Ready Transition, Medium Might Have Changed`. Some SD card readers report this on power on. I assume the same thing could happen with USB disks.

The problem is that we simply don't know what to do to resolve it. What the correct sequence of messages to the USB device is for it to stop reporting this issue. Perhaps it's not even possible to fix but a hardware issue. I can imagine inserting and removing the disk repeatedly can give it issues. Perhaps wait longer when you remove it so it has time to power down completely?

MagicHuHuHu, 2022-03-14:
I had the same problem, according to @Derpalus comment. I did a driver repair on the USB stick on my computer and the problem gone.

MagicHuHuHu, 2022-03-15:
> I had the same problem, according to @Derpalus comment. I did a driver repair on the USB stick on my computer and the problem gone.
I try another machine.It's show the error again.But if I restart the application,it can worked.