magnusja/libaums #284: OutOfMemoryError FAT.getChain()

ioapps-dev, 2021-01-30:
This has happened on some devices, I have no further information at the moment.

java.lang.OutOfMemoryError: 
  at java.util.Arrays.copyOf (Arrays.java:3139)
  at java.util.Arrays.copyOf (Arrays.java:3109)
  at java.util.ArrayList.grow (ArrayList.java:275)
  at java.util.ArrayList.ensureExplicitCapacity (ArrayList.java:249)
  at java.util.ArrayList.ensureCapacityInternal (ArrayList.java:241)
  at java.util.ArrayList.add (ArrayList.java:467)
  at com.github.mjdev.libaums.fs.fat32.FAT.getChain$libaums_release (SourceFile:6)
  at com.github.mjdev.libaums.fs.fat32.ClusterChain.<init> (SourceFile:2)
  at com.github.mjdev.libaums.fs.fat32.FatDirectory$Companion.readRoot$libaums_release (SourceFile:2)
  at com.github.mjdev.libaums.fs.fat32.Fat32FileSystem.<init> (SourceFile:7)
  at com.github.mjdev.libaums.fs.fat32.Fat32FileSystem.<init> (SourceFile:1)
  at com.github.mjdev.libaums.fs.fat32.Fat32FileSystem$Companion.read (SourceFile:12)
  at com.github.mjdev.libaums.fs.fat32.Fat32FileSystemCreator.read (SourceFile:1)
  at com.github.mjdev.libaums.fs.FileSystemFactory.createFileSystem (SourceFile:2)
  at com.github.mjdev.libaums.partition.Partition$Companion.createPartition (SourceFile:2)
  at com.github.mjdev.libaums.UsbMassStorageDevice.initPartitions (SourceFile:5)
  at com.github.mjdev.libaums.UsbMassStorageDevice.setupDevice (SourceFile:14)
  at com.github.mjdev.libaums.UsbMassStorageDevice.init (SourceFile:2)

I think on some devices this loop never ends.

FAT.kt
do {
            result.add(currentCluster)
            offset = (fatOffset[0] + currentCluster * 4) / bufferSize * bufferSize
            offsetInBlock = (fatOffset[0] + currentCluster * 4) % bufferSize

            // if we have a new offset we are forced to read again
            if (lastOffset != offset) {
                buffer.clear()
                blockDevice.read(offset, buffer)
                lastOffset = offset
            }

            currentCluster = (buffer.getInt(offsetInBlock.toInt()) and 0x0FFFFFFF).toLong()
        } while (currentCluster < FAT32_EOF_CLUSTER)

magnusja, 2021-08-30:
Seems that there is no end marker in the cluster chain
Might be related to #247

magnusja, 2023-03-11:
Fixed with v0.9.4

closed 2023-03-11 by magnusja