magnusja/libaums #292: Transfer large file always error

lzxshare, 2021-04-24:
First, thanks for this grate lib and also thanks for grate author,i think it's very useful to many developers, and support a very good solution on USB transfer in Android platform.

I have a problem when i use this lib to copy phone externalStorage files to usb device. Normally,transfer small file  works very good. but when i transfer large file, almost 3GB,it always make mistakes. My usb device is Fat32 FileSystem,i know the Fat32 FileSystem can not transfer single file more than 4GB. I am not sure that  if this limitation  caused that problem. Had you tested transferring more than 4GB files? . Here is the error log:

java.io.IOException: MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device 
    and try again
    at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:181)
    at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.write(ScsiBlockDevice.kt:340)
    at com.github.mjdev.libaums.driver.ByteBlockDevice.write(ByteBlockDevice.kt:104)
    at com.github.mjdev.libaums.fs.fat32.ClusterChain.write$libaums_release(ClusterChain.kt:232)
    at com.github.mjdev.libaums.fs.fat32.FatFile.write(FatFile.kt:113)
    at com.github.mjdev.libaums.fs.UsbFileOutputStream.write(UsbFileOutputStream.kt:75)
    at java.io.BufferedOutputStream.write(BufferedOutputStream.java:122)
    at okio.OutputStreamSink.write(JvmOkio.kt:53)
    at okio.RealBufferedSink.emit(RealBufferedSink.kt:261)
    at com.xxx.xxxxx.BackupService$BackupServiceTask.backupAction(BackupService.java:131)
    at com.xxx.xxxxx.BackupService$BackupServiceTask.executeBackupWork(BackupService.java:189)
    atcom.xxx.xxxxx.local.BackupService.onHandleWork(BackupService.java:58)
    at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:391)
    at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:382)
    at android.os.AsyncTask$3.call(AsyncTask.java:394)
    at java.util.concurrent.FutureTask.run(FutureTask.java:266)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
    at java.lang.Thread.run(Thread.java:923)
ice: Could not read from device, result == -1 errno 0 null, retrying...
ice: Could not read from device, result == -1 errno 0 null, retrying...
10276(com.veb.privatespace) AsyncTask #1 identical 1 line

Could you help me analyze what causes it? and than, How can I solve this problem? Could you give me some positive ideas? I'm 

in a hurry to deal with this problem. Thank you very much!

Depau, 2021-04-24:
You're most likely wiggling the adapter while it's transferring. Try to keep it still on a table.

lzxshare, 2021-04-24:
so，i think you get me wrong about this issue. i had tested many times, my usb device connected phone well. it always appeared when transfer 70~90 percent. but when test it with small files, it works well.  what my mean is  that when use this lib to transfer large files, should  i do something else settings in the code. as you know,transfer large file can cost much time and resources, it's important to promise codes stability. so,i want to know,had ever you tested transfering large files? whether it care about usb device when transfer large files?

Depau, 2021-04-24:
> so，i think you get me wrong about this issue. i had tested many times, my usb device connected phone well. it always appeared when transfer 70~90 percent. but when test it with small files, it works [...]

Yes.

https://github.com/magnusja/libaums/blob/04badb70752dce1614e24f104136e0e50b1e87d8/libaums/src/main/java/com/github/mjdev/libaums/driver/scsi/ScsiBlockDevice.kt#L163-L178

This is the code that triggers the error. If it fails there it's most likely an hardware error, and being this a userspace library there's so much you can do to recover the transfer.

On top of that, Android's permission model doesn't allow regaining access to the device without reconnecting it and re-granting the permissions.

> transfer large file can cost much time and resources, it's important to promise codes stability

If this is a concern for you and if big file transfer failures causes actual costs to you, you should implement resuming transfers on reconnection instead of letting one random library be your single point of failure in your important business logic. There's little to no buffering so you can easily resume the write of a single file without restarting it. After all, do you really want cheap wobbly OTG dongles to put  your transfer at risk?

lzxshare, 2021-04-24:
Ok，I appreciate your patient explanation so much. but i still think you don't see my point. ha-ha，maybe i don't described my question clearly. I'm not good at English.  in fact,resuming transfers can not solve my problem. I've been debugging for this problem full today afternoon,the result  could be much more serious. for example,after transfer failed, the file is damaged on usb device ,and still taking up  usb device space,although i write the delete UsbFIle code when it transfer failed,it seems not work. in the end，USB communication on Android platform is  a  very headache problem, you can hardly find any valuable information in the Google official documents. I don't have much time to study the complete USB communication technology. What I really care about is whether i got a wrong way when i use your lib or there is truly some problems when transfer large file with this lib. I want to solve this problem from the root. so i ask you had you been tested transferring single file more than 4GB? this answer is very important to me. I'm also Researching your project and  learn your demo code, i'm trying  to look for some different  details between your demo  and my code. this problem had troubled me a few days, i am crashing about it. i really hope i can kill this problem tomorrow.I also hope I can  discuss this issue go on with you. after all, you are the author of this lib,you're familiar with the code.  thank you again.

Depau, 2021-04-24:
> you can hardly find any valuable information in the Google official documents

Yeah, in fact we ditched it altogether eventually (if your app's license allows LGPL code the `libusb`-based implementation should always be preferred)

In general, this is hardware issues. On real hardware it's very hard to test, since you need a physical device, connect it to a USB drive and wait for the entire transfer to finish.

I spent lots of time to implement automated testing on QEMU and it's already quite hit or miss on CI platforms - on top of that they would only run on Travis and Travis cut down on free open source software accounts last year. https://github.com/magnusja/libaums/tree/ci-instrumented-tests

The Android storage access framework will most likely work better than this if you're okay setting the min SDK version to a high-ish number.

We can't review whether you're using the library correctly if you only post the stack trace and no code snippets.

lzxshare, 2021-04-25:
Hi,have a nice day. today,i took  a full day to investigate the problem with transfer large file. i am so tired ha-ha.  Luckily， i found the problem in finally. the whole process was so long but the result just like a joke. there is no problem with your lib and also no problem on my way to use. It's a very secretive question，what cause that problem is  the limitations of the Android system. 

the flow of my whole function is packing the external storage file and encrypt them and compressed,than transfer these to USB Device. it sounds very complicated. Without doubt，it should be running in the background task. so I use the JobIntentService API to implement the function,that's the problem on it. packing files、encryption and transfer to USB Device can cost much time. the JobIntentService can not running more than ten minutes, i got this information from [https://stackoverflow.com/] website. i am not sure ten minutes is an accurate limitation time about JobIntentService,but i am sure that JobIntentService  dose not give the background task much time to running. that's a good explanation for why there's always be interrupted when transfer 70~90 percent.

In the process of debugging，i also find some problems else. limited by Fat32 FileSystem,we can not transfer single file more 4GB, it's an obvious problem in transfering files,i think you should mention it on readme. I also changed your source code in my project, the java IO Stream has an lower  performance，so i used okio to read and write files. most like this:
[image]

do you think it's necessary to do this? 

Anyway，I finally solved the problem which made me headache. Yeah，I had to complain about Google had done some changes on Android system，especially after Android 8.0. many developers are disappointed with some system level restrictions. it makes developer hard to develop some function.

i'm honoured to be a technology developer as you. i appreciate your help these days very much, and so many developers  will be  grateful about this great lib and great open source Author.

Good luck! see you.

lzxshare, 2021-09-08:
it is my fault

closed 2021-09-08 by lzxshare