magnusja/libaums #293: Does not work on Samsung w/ OneUI 3.0 / Android 11

Depau, 2021-04-27:
Forwarding here from https://github.com/EtchDroid/EtchDroid/issues/117

### Problem

The affected users seem to be running Samsung's downstream ROM "OneUI" version 3.0 or later. I was able to borrow a devices with OneUI 1.0 and it doesn't seem to be affected.

I'm currently testing on a device with OneUI 2.1, though I'm having issues with wireless ADB and I will have to return it soon.

### Expected behavior

### Actual behavior

No user provided a logcat but the behavior seems the same as when the app is closed, then re-started with the same USB drive selected without unplugging and re-plugging the USB drive (ANR then timeout)

#### Stacktrace of Excpetion

```
(if exists)
```

### Code where problem occurs

magnusja, 2021-05-12:
Interesting, I read the thread in EtchDroid/EtchDroid#117 but I am as confused as you probably are. 

I am not sure if we are missing something to make this fully compatible with Android 11?

I am assuming you are using the libusb communication? Maybe you want to package an app for the guys in EtchDroid/EtchDroid#117 using the vanilla Android USB API and see if that makes any difference?

Depau, 2021-05-13:
> I am assuming you are using the libusb communication? Maybe you want to package an app for the guys in EtchDroid/EtchDroid#117 using the vanilla Android USB API and see if that makes any difference?

That's not a bad idea actually, I didn't think about it. No one provided logs though, it's hard to tell what's breaking exactly.

I just posted this here hoping you have or know someone who has a recent Samsung, it's hard to borrow devices around these times :)

shirt-dev, 2021-08-02:
> That's not a bad idea actually, I didn't think about it. No one provided logs though, it's hard to tell what's breaking exactly.
>
> I just posted this here hoping you have or know someone who has a [...]

It recently worked for me on my Samsung Galaxy S20 FE running OneUI 3.0 with no issues. If you need any testing let me know.

Steve-George, 2021-08-03:
I am using MIUI in Redmi Note 10 Pro. This software cannot function well, too. It always reports an error and then screen will turn to    
black.

Moohee-yll, 2021-11-19:
I was very troubled with the same problem. I found this is caused by function:  claimInterface (UsbInterface intf, **boolean force**). 

- If I let force == true (just like in libaums), then the crash appeared. 
- If I let force == false, this problem disappeared but the return value of  "claimInterface" is false, so I was unable to continue the next step.

So I this the key point is claimInterface( ).But I don't know how to do.

steroidraging, 2022-07-11:
Are you trying this with or without OTG storage enabled in settings? 

Something I noticed, using my OnePlus Nord N10 5G is that for some reason with that enabled as you would typically need for the phone to notice the drive right, it actually throws off EtchDroid on my device. 

One day I tried EtchDroid and actually had forgotten to enable that option in System Settings and even tho most apps won't even recognize the drive without that enabled, it was the ONE thing that actually helped EtchDroid to fully complete the writing of the img successfully.

depau, 2025-05-02:
@steroidraging I just noticed your reply. This is interesting, do you still have that device? I've never seen any phone with such an option.

depau, 2026-04-05:
@Ben-bit-code208 if you open a PR feel free to add me as a reviewer, I can take a look

mikenorthern96-stack, 2026-08-17:
I was banned from etch droid by git hub after they wrote the ISO with the excuse of it's above their pay grade

mikenorthern96-stack, 2026-08-17:
If indeed you have everything need to complete this which now that I've seen the problem I'm not sure because in the etchdroid had only written part of it and like Rufus it doesn't notify of problems and skips important partsand you end up mad