magnusja/libaums #297: Received status code 403 from server: Forbidden

JAICHANGPARK, 2021-05-24:
I had some problem build & run develop branch code

```
Execution failed for task ':app:mergeDebugResources'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve com.github.magnusja:java-fs:0.1.5.
     Required by:
         project :app > project :javafs
      > Could not resolve com.github.magnusja:java-fs:0.1.5.
         > Could not get resource 'https://dl.bintray.com/magnusja/maven/com/github/magnusja/java-fs/0.1.5/java-fs-0.1.5.pom'.
            > Could not GET 'https://dl.bintray.com/magnusja/maven/com/github/magnusja/java-fs/0.1.5/java-fs-0.1.5.pom'. Received status code 403 from server: Forbidden

```

danielmeza, 2021-06-01:
I get this error too, I'ts caused by the disposition of bintray.

magnusja, 2021-06-01:
Hey there,

sorry the news that bintray is shutting down reached me only recently. 

For now, please checkout the repo and compile for yourself. I will switch to maven again, but not sure how long this will take, the reason I used bintray is that was much easier. ..

danielmeza, 2021-06-01:
I have switched but the javafs project has a dependency to java-fs:0.1.5 and can't get that library.

magnusja, 2021-06-01:
If you dont need java-fs then just exclude it, otherwise you can get the source code here: https://github.com/magnusja/java-fs

magnusja, 2021-06-28:
Fixed on `develop`

closed 2021-06-28 by magnusja