magnusja/libaums #302: Unable to read exFAT and NTFS USB drives

prathameshkulkarni12, 2021-06-15:
While reading the files for exFAT or NTFS USB drives I am getting the following error.

java.lang.RuntimeException: Error receiving broadcast Intent { act=com.android.example.USB_PERMISSION flg=0x10 (has extras) } in com.eits.dataupload.Activities.MainActivity$1@a35c818
at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1566)
at android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run(Unknown Source:2)
at android.os.Handler.handleCallback(Handler.java:883)
at android.os.Handler.dispatchMessage(Handler.java:100)
at android.os.Looper.loop(Looper.java:224)
at android.app.ActivityThread.main(ActivityThread.java:7592)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:539)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)
Caused by: java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
at java.util.ArrayList.get(ArrayList.java:437)
at com.eits.dataupload.Activities.MainActivity$1.onReceive(MainActivity.java:336)
at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1556)

magnusja, 2021-06-28:
Code?

prathameshkulkarni12, 2021-06-29:
> Code?

Using your code directly I have not modified anything in the code

prathameshkulkarni12, 2021-06-29:
> Code?

`UsbMassStorageDevice[] devices = UsbMassStorageDevice.getMassStorageDevices(getActivity());

        if (devices.length > 0)
            mSelectedDevice = devices[0];

            mSelectedDevice.init();

            // we always use the first partition of the device
            FileSystem fs = mSelectedDevice.getPartitions().get(0).getFileSystem();
            UsbFile root = fs.getRootDirectory();`

It is crashing on the FileSystem line.
I used another Pendrive then it is working, but if I use exFAT or NTFS format pen drive, it is giving the mentioned error.

magnusja, 2021-06-29:
Did you include the `java-fs` module?That is necessary for other file systems than fat32

prathameshkulkarni12, 2021-06-29:
No, I have not used that. is that included in your library or it is another module that I have to include? Is so then can you give me some reference how to do that.

magnusja, 2021-06-29:
It is highly experimental that is why it is not documented that well. But there is a folder called `java-fs` which you need to include into your android app as a module.

The example app actually uses it https://github.com/magnusja/libaums/blob/develop/app/src/main/java/com/github/mjdev/libaums/usbfileman/MainActivity.kt#L80

closed 2021-06-29 by magnusja

cgapk123, 2025-07-06:
> It is highly experimental that is why it is not documented that well. But there is a folder called `java-fs` which you need to include into your android app as a module.
>
> The example app actually [...]

404 - page not found
There is no more sample code for using java-fs. Is there any other sample code?