magnusja/libaums #309: Maven publishing is not working

lennynilsson, 2021-07-13:
### Problem
Bintray has been shut down and therefore no artifact will be published.
There inconsistencies in the Maven Central repository which prevents any artifact to be downloaded.
Eather 

### Expected behavior
Pom files and artifacts should be available on Maven Central for new releases

### Actual behavior
No pom file or artifact are available on Maven Central

#### Stacktrace of Excpetion

[https://travis-ci.org/github/magnusja/libaums](https://travis-ci.org/github/magnusja/libaums)
[https://bintray.com/magnusja/maven/libaums/_latestVersion](https://bintray.com/magnusja/maven/libaums/_latestVersion)

### Code where problem occurs

[https://github.com/magnusja/libaums/blob/master/build.gradle](https://github.com/magnusja/libaums/blob/master/build.gradle)

lennynilsson, 2021-07-13:
Hi @magnusja, I'm sorry to bring you yet another issue. On behalf of my company I would like to thank you for the contributions you've done thought this library.

Let me know if there is anything I can do to help in order to resolve this publishing issue.

magnusja, 2021-09-29:
See https://github.com/magnusja/libaums/issues/315#issuecomment-930520993

closed 2022-01-22 by magnusja