magnusja/libaums #322: How can I use PackageManager.getPackageArchiveInfo on USB file

xulifan, 2021-11-01:
### Problem
I am developing virus scan application. I need scan the files inside USB drive. With libaums, I can list all the files inside a USB drive. However, to parse an APK file using PackageManager.getPackageArchiveInfo, I need the full path of the file. How can I do this with libsums?

### Expected behavior

### Actual behavior

#### Stacktrace of Excpetion

```
(if exists)
```

### Code where problem occurs

magnusja, 2022-01-22:
Not possible as this is a user space lib. You need to parse the apk yourself or see if it offers an API which does not take file path but instead raw bytes.

magnusja, 2022-01-22:
maybe the storageprovider module can help but I am unsure

closed 2022-01-22 by magnusja