magnusja/libaums #323: Getting IndexOutOfBoundsException even on FAT32 USB while getting device Partitions.

rahulshyokand, 2021-11-08:
### Problem

Getting `IndexOutOfBoundsException` even on FAT32 USB.

#### Is there any specific way to format USBs to FAT32 to be able to get partitions access?
The USB works well with other apps and devices as we would expect.

### Expected behavior
This HP USB is formatted various times with FAT32 and should be able to access partitions as working on many USBs.

### Actual behavior

```
W/Partition: Unsupported fs on partition
W/System.err: java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
```

[image]

#### Stacktrace of Excpetion

USB Device Detected
```

2021-11-08 19:20:16.195 30015-30015 I/UsbMassStorageDevice: found usb device: /dev/bus/usb/001/002=UsbDevice[mName=/dev/bus/usb/001/002,mVendorId=1008,mProductId=32064,mClass=0,mSubclass=0,mProtocol=0,mManufacturerName=HP,mProductName=v236w,mVersion=1.00,mSerialNumberReader=android.hardware.usb.IUsbSerialReader$Stub$Proxy@6ce7e46, mHasAudioPlayback=false, mHasAudioCapture=false, mHasMidi=false, mHasVideoCapture=false, mHasVideoPlayback=false, mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=100,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=8,mSubclass=6,mProtocol=80,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=2,mMaxPacketSize=512,mInterval=0]
    UsbEndpoint[mAddress=2,mAttributes=2,mMaxPacketSize=512,mInterval=0]]]]
2021-11-08 19:20:16.195 30015-30015 I/UsbMassStorageDevice: Found usb interface: UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=8,mSubclass=6,mProtocol=80,mEndpoints=[
    UsbEndpoint[mAddress=129,mAttributes=2,mMaxPacketSize=512,mInterval=0]
    UsbEndpoint[mAddress=2,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
```

Exception Event
```
2021-11-08 19:20:15.517 30015-30015/ I/UsbMassStorageDevice: Found usb endpoint: UsbEndpoint[mAddress=129,mAttributes=2,mMaxPacketSize=512,mInterval=0]
2021-11-08 19:20:15.517 30015-30015/ I/UsbMassStorageDevice: Found usb endpoint: UsbEndpoint[mAddress=2,mAttributes=2,mMaxPacketSize=512,mInterval=0]

2021-11-08 19:20:15.523 30015-30015 I/UsbMassStorageDevice: MAX LUN 0
2021-11-08 19:20:15.524 30015-30015 D/ScsiBlockDevice: inquiry response: ScsiInquiryResponse [peripheralQualifier=0, peripheralDeviceType=0, removableMedia=false, spcVersion=6, responseDataFormat=2]
2021-11-08 19:20:15.528 30015-30015/ I/ScsiBlockDevice: Block size: 512
2021-11-08 19:20:15.528 30015-30015/ I/ScsiBlockDevice: Last block address: 30285823
2021-11-08 19:20:15.530 30015-30015/ D/MasterBootRecord: Unknown partition type-18
2021-11-08 19:20:15.531 30015-30015/ W/Partition: Unsupported fs on partition

2021-11-08 19:20:15.553 30015-30015/ W/System.err: java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
2021-11-08 19:20:15.553 30015-30015/ W/System.err:     at java.util.ArrayList.get(ArrayList.java:437)
```

### Code where the problem occurs
```
 // Only uses the first partition on the device
FileSystem currentFs = device.getPartitions().get(0).getFileSystem();
```

Getting the same exception on both versions of libaums library.
``` 
implementation 'com.github.mjdev:libaums:0.7.4'

implementation 'me.jahnen:libaums:0.8.0'
```

vasiledoe, 2021-12-06:
I had the same issue but my problem was a 256GB SD card exFAT instead of FAT32. Just formatted it to FAT32 and problem was gone.

magnusja, 2021-12-07:
Are you sure that this is using MBR and not GPT?

rahulshyokand, 2021-12-08:
> Are you sure that this is using MBR and not GPT?
This might be the issue I faced.
The same USB was used to create bootable for Windows a few days back but even after the format to FAT32, it didn't work.
As you rightly pointed I think the issue may be about MBR and GPT.
I didn't know the difference between the two.
It would be better if we can write the difference between MBR and GPT in the documentation somewhere for avoiding confusion for future lib users.
Thanks

closed 2022-01-22 by magnusja

talhacomak, 2022-09-21:
I faced the same issue too. I changed the usb GPT to MBR then the problem resolved. I guess it would be better if you find a solution for GPT types or emphasize it in documentation.

Thanks.

magnusja, 2023-03-11:
GPT is now supported!