magnusja/libaums #324: Using Libaums just to measure free drive space and then close it

gavingt, 2021-11-18:
Libaums has very fast methods for measuring free drive space and drive capacity, whereas Android 12 has crippled our ability to measure these through normal means. My goal is to copy files to the USB drive using the standard Android API and Uris, and then when I'm done I want to use Libaums to measure the drive space. But I'm running into a problem with this, as it seems that Libaums won't fully release control even after I call `usbMassStorageDevice.close()`. In fact, using it corrupts many of the files that were just copied to the USB drive. Is there some way to use Libaums side-by-side with the standard Android APIs?

Here's my method for measuring space:

```
fun getDriveInfoFromLibaums(): Boolean {
    val deviceListFromLibaums = UsbMassStorageDevice.getMassStorageDevices(appContext)
    if (deviceListFromLibaums.isNullOrEmpty()) return false

    var usbMassStorageDevice: UsbMassStorageDevice? = null
    try {
        usbMassStorageDevice = deviceListFromLibaums[0]
        val tempUsbDevice = usbMassStorageDevice.usbDevice
        usbMassStorageDevice.init()
        val currentFileSystem = usbMassStorageDevice.partitions[0].fileSystem

        // This creates a file on the root directory of the USB drive, which we then read later to
        // verify that the user selected the root directory of the USB drive from the SAF picker.
        try {
            currentFileSystem.rootDirectory.createFile(FILE_NAME_VALIDATE_USER_SELECTED_ROOT_DIRECTORY)
        } catch (throwable: Throwable) {
            // validation file already exists, so we swallow the error.
        }

        prefs.lastRecordedSerialNumber = tempUsbDevice.serialNumber ?: ""
        prefs.lastRecordedDriveSize = currentFileSystem.capacity
        prefs.lastRecordedDriveFreeSpace = currentFileSystem.freeSpace
        usbMassStorageDevice.close()

        // Updates phone home data.
        PhoneHomeData.instance.setFloatValue(PhoneHomeData.kPhoneHomeDriveSizeKey, bytesToGB(prefs.lastRecordedDriveSize))
        return true
    } catch (throwable: Throwable) {
        logMessage("Tools", "Error occurred in getDriveInfoFromLibaums(): ${throwable.message}")
        usbMassStorageDevice?.close()
        return false
    }
}
```

Depau, 2021-11-19:
The Android storage access framework will use the regular filesystem to write data, which means it will do some buffering and finish writing in the background.

If you want to take control of the USB drive with libaums you need to ensure the drive is "safely removed" by somehow invoking the "Eject" functionality that is found in the system USB notification. I don't know how you can do this programmatically.

gavingt, 2021-11-19:
Thanks for the response. This certainly explains the problems I've been experiencing. I've found that I can go from Libaums to SAF with no problem, but going SAF to Libaums results in corrupted files. 

I've tried to look through the Libaums code to learn how it calculates drive space. It seems like the approach is similar to [this](https://stackoverflow.com/questions/70018379/contentresolver-openassetfiledescriptor-takes-a-long-time-in-android-12) except it's operating on custom classes and it's much faster in Android 12. Is there perhaps a way to use the Libaums approach within the standard Android APIs?

Depau, 2021-11-19:
> Is there perhaps a way to use the Libaums approach within the standard Android APIs?

No. Libaums directly accesses the USB drive by issuing SCSI commands directly, bypassing the Linux kernel's block device subsystem. Android does not, and it should never, since it is not in the condition of a regular unprivileged app that needs to work around limitations.

You should probably look for the source code of `ContentResolver.openAssetFileDescriptor()`, see what it does and see what changed in the latest release.

gavingt, 2021-11-19:
Can you confirm that Libaums doesn't do any such buffering or writing in the background after calling `usbMassStorageDevice.close()`? I'm trying to determine whether it's safe to use Libaums at app start to measure drive space and then close it and use SAF after.

Depau, 2021-11-20:
It does not. However, yours doesn't look like a great idea: there could be pending writes *before* the user launches your app and you're likely to corrupt their filesystem.

How about tweaking the user experience so the user doesn't get bored while you're loading the disk space? Add a spinner and wait for the file descriptor in a background thread.

closed 2021-11-20 by Depau