magnusja/libaums #325: 因为无法获取真实路径，我应该如何使用libaums安装U盘中的apk，copy？不能直接安装吗？

FlyFelix, 2021-12-07:
### Problem

### Expected behavior

### Actual behavior

#### Stacktrace of Excpetion

```
(if exists)
```

### Code where problem occurs

zongren, 2021-12-09:
是的，复制到内部存储然后使用FileProvider获取Uri

FlyFelix, 2021-12-09:
好的，谢谢

FlyFelix, 2021-12-09:
好的谢谢

 

 
&mdash;

656643899, 2022-01-14:
可直接在u盘安装apk!!!
系统安装方法实际也是先通过读写IO流copy apk,那么我们也可以通过new UsbFileInputStream(UsbFile)获取到IO流，再在系统方法基础上替换获取IO流的部分即可.

代码如下：
    @RequiresApi(api = Build.VERSION_CODES.LOLLIPOP)
    public static void install28UsbFile(Context context, UsbFile apkFile,InstallCallback callback) {
        PackageInstaller packageInstaller = context.getPackageManager().getPackageInstaller();
        PackageInstaller.SessionParams sessionParams = new PackageInstaller.SessionParams(PackageInstaller.SessionParams.MODE_FULL_INSTALL);
        sessionParams.setSize(apkFile.getLength());

        int sessionId = createSession(packageInstaller, sessionParams);
        if (sessionId != -1) {
            boolean copySuccess = copyInstallUsbFile(packageInstaller, sessionId, apkFile);
            if (copySuccess) {
                execInstallCommand(context, packageInstaller, sessionId,apkFile.getName());
            }else {
                if (callback !=null)
                    callback.onFailure(apkFile.getName());
            }
        }else {
            if (callback !=null)
                callback.onFailure(apkFile.getName());
        }
    }

    @RequiresApi(api = Build.VERSION_CODES.LOLLIPOP)
    private static boolean copyInstallUsbFile(PackageInstaller packageInstaller,
                                           int sessionId, UsbFile apkFile) {
        InputStream in = null;
        OutputStream out = null;
        PackageInstaller.Session session = null;
        boolean success = false;
        try {
            session = packageInstaller.openSession(sessionId);
            out = session.openWrite("base.apk", 0, apkFile.getLength());
            in = new UsbFileInputStream(apkFile);
            int total = 0, c;
            byte[] buffer = new byte[65536];
            while ((c = in.read(buffer)) != -1) {
                total += c;
                out.write(buffer, 0, c);
            }
            session.fsync(out);
            Log.d(TAG, "streamed " + total + " bytes");
            success = true;
        } catch (IOException e) {
            e.printStackTrace();
        } finally {
            closeQuietly(out);
            closeQuietly(in);
            closeQuietly(session);
        }
        return success;
    }

    @RequiresApi(api = Build.VERSION_CODES.LOLLIPOP)
    private static void execInstallCommand(Context context, PackageInstaller packageInstaller, int sessionId,String apkFilePath) {
        PackageInstaller.Session session = null;
        try {
            session = packageInstaller.openSession(sessionId);
            Intent intent = new Intent(context, InstallResultReceiver.class);
            intent.putExtra("apkFilePath",apkFilePath);
            PendingIntent pendingIntent = PendingIntent.getBroadcast(context, 1, intent, PendingIntent.FLAG_UPDATE_CURRENT);
            session.commit(pendingIntent.getIntentSender());
        } catch (IOException e) {
            e.printStackTrace();
        } finally {
            closeQuietly(session);
        }
    }

zongren, 2022-01-15:
> 可直接在u盘安装apk!!! 系统安装方法实际也是先通过读写IO流copy apk,那么我们也可以通过new UsbFileInputStream(UsbFile)获取到IO流，再在系统方法基础上替换获取IO流的部分即可.
>
> 代码如下： @RequiresApi(api = Build.VERSION_CODES.LOLLIPOP) public static void [...]

这种方式还是复制到临时文件，然后再安装吧？

FlyFelix, 2022-01-15:
谢谢，这种方法是可行的，十分感谢

zongren, 2022-01-15:
@FlyFelix 请问打开速度有提升吗

closed 2023-03-11 by magnusja