magnusja/libaums #326: Cannot read all files from USB drive

vasiledoe, 2021-12-07:
### Problem
Not all files are returned when we try to read usb via `listFiles()`. I have a usb with 1000+ pictures and this method return only a part of them. For different size files it returns a different amount so I guess this might be related to some buffer size or memory related stuff.
I tried this with different USB and SD cards(connected via OTG adapter) on different Andorid 11 devices.

### Code where problem occurs
```
val currentFs = storageDevice.partitions[0].fileSystem
val rootDir = fileSystem.rootDirectory
val files = rootDir.listFiles()
```

vasiledoe, 2021-12-08:
@magnusja any idea about this one? 
I represented the basic code, that's in docs.

magnusja, 2021-12-08:
Unfortunately not, I would need to prepare a test case for this first and see if I can reproduce this.

vasiledoe, 2022-02-01:
@magnusja can I help? 
I investigated in debug and it seems to read not all `entries` in `FatDirectory` via `readEntries()` method. Not sure how do you read files, based on what chunking methodology but I assume it reads only from one block/sector/whatever you call it.

magnusja, 2022-02-08:
Do you think it could be related to https://github.com/magnusja/libaums/issues/298

vasiledoe, 2022-03-15:
just downloaded the whole project but cannot run in Android Studio due to errors:
```
C/C++: C:\Users\Pro\Desktop\testusblib\libusbcommunication\CMakeLists.txt debug|x86 : CMake Error at C:\Users\Pro\Desktop\testusblib\libusbcommunication\CMakeLists.txt:11 (add_library):
  Cannot find source file:

    null/libusb/core.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx
```
Not sure how can I test your sugestion other then downloading the project and checkout that branch.

magnusja, 2022-03-15:
You need to download libusb 1.0.25 and set `libusb.dir` to the path where you extracted the source tarball. See gitlab ci for more info. https://github.com/magnusja/libaums/blob/566a5a9c8dd64a32fc1627b5d7de383d79abe12d/.github/workflows/android.yml#L16-L19

vasiledoe, 2022-03-17:
I got v1.0.25 for libusb and also seems to work. So the main issue why not all files were disovered is because of missing this:
```
FileSystemFactory.registerFileSystem(JavaFsFileSystemCreator())
UsbCommunicationFactory.registerCommunication(LibusbCommunicationCreator())
UsbCommunicationFactory.underlyingUsbCommunication = UsbCommunicationFactory.UnderlyingUsbCommunication.OTHER
```
Even in your demo code, if we comment these lines (`MainActivity.init()`) you can see it reads only couple files from the USB. 
Unfortunately this is no mentioned on [lib lander page docs here on github](https://github.com/magnusja/libaums).
Also we canot use them only via `implementation 'me.jahnen.libaums:core:0.9.1'` because there is no dependency.

magnusja, 2022-03-17:
1.0.25 is supported as well yes.

So yeah what you are basically saying there is a bug in my fat32 implementation, that is what I suspected :D

As it is a bug, and this should work, or at least I intent to provide a working fat32 implementation I am not mentioning it in the readme.

What do you mean there is no dependency?

vasiledoe, 2022-03-17:
I mean if we import your lib only via gradle `implementation 'me.jahnen.libaums:core:0.9.1'` then we cannot use:
```
FileSystemFactory.registerFileSystem(JavaFsFileSystemCreator())
UsbCommunicationFactory.registerCommunication(LibusbCommunicationCreator())
UsbCommunicationFactory.underlyingUsbCommunication = UsbCommunicationFactory.UnderlyingUsbCommunication.OTHER
```

because can't find dependency to these classes. On the other hand, if we import `libusbcommunication`, `javafs` and `libaums` as modules, and add `implementation project(path: ':MODULE_NAME')` for each of them in gradle then these lines of code can find their dependencies and everithing works as expected reading all files on USB cards. 
I assume you published only `libaums` to  `me.jahnen.libaums:core:0.9.1` but `libusbcommunication` and `javafs` is missing so that's why those lines of code can't see depndencies.

vasiledoe, 2022-03-17:
So the point is to mention in "How to use" docs about REQUIREMENT of using this init code lines:
```
FileSystemFactory.registerFileSystem(JavaFsFileSystemCreator())
UsbCommunicationFactory.registerCommunication(LibusbCommunicationCreator())
UsbCommunicationFactory.underlyingUsbCommunication = UsbCommunicationFactory.UnderlyingUsbCommunication.OTHER

```
and at the same time to incluse all dependencies in published lib

magnusja, 2022-03-17:
This is 100% optional.

With that you are including two LGPL libraries, namely JNode OS and libusb. The libaums core is apache only.

This library is structured in modules, see the readme for installing additional modules via dependency.

https://github.com/magnusja/libaums/tree/develop/libusbcommunication#inclusion-in-your-buildgradle

The reason java fs is not published on maven is that it is highly experimental and not suited for production.

vasiledoe, 2022-03-17:
Do you have any clue why this init code can apparently solve the issue? I mean, can this be considered to solve the issue or we need to focus on internal read implementation and solve that bug instead?

magnusja, 2022-03-17:
As mentioned earlier, you are using the JNode OS file system implementation. This implementation does not have the same bug as my fat32 implementation has. I dont think using libusbcommunication changes anything regarding the issue you are having.

I would not consider this solved, as my implementation is not functioning properly. Main incentive is that JNode does not comply with apache, and thus limits the use in closed source applications. Some people even say that it is impossible to have a closed source application using any LGPL libraries, thus you need to publish your source code. See here https://xebia.com/the-lgpl-on-android/

If this solves the issue for you, I cannot tell, you are the better person to judge if this is acceptable within your use case.

Mahmoud-Mustafa-Dev, 2022-04-16:
@magnusja  @vasiledoe Anything new with this issue? I am facing the same thing, I have 1k images and the listFiles() is only returning 300 of them

closed 2023-03-11 by magnusja