magnusja/libaums #327: UsbFile#delete corrupts FAT16 files

meguro-pgv, 2021-12-14:
### Problem
After deleting a file, the other files get corrupted in certain cases.
It looks similar to the issue https://github.com/magnusja/libaums/issues/144.
I will attach memory images of the USB memory before and after deleting a file.
(before_deleting_1st_file.img and after_deleting_1st_file.img.
 For attachment, each file was split by 10MB and zipped.)

I am using a USB memory that is under development by our company.
Filesystem : FAT16
Memory capacity : 128 MB
Sector size (BPB_BytsPerSec) : 4096 Bytes
Number of sectors per cluster (BPB_SecPerClus) : 1

I am using libaums:0.7.4 and java-fs:0.1.4 in AndroidStudio.
< build.gradle >
```
implementation 'com.github.mjdev:libaums:0.7.4'
implementation 'me.jahnen:java-fs:0.1.4'
```

I tried the latest code in [the master branch of Nuix/Jnode-fs (commit : 72d3e97)](https://github.com/Nuix/jnode-fs/commit/72d3e97e7104e6aa3778d8121132db0ea441252c) and the problem reproduced yet.

### Expected behavior
After deleting a file, the remaining files are not corrupted and can be read correctly.

### Actual behavior
There are 8 files in the USB memory.
When I delete any one of the 1st to 6th files, the 7th and 8th files get corrupted (FAT entries are set to 0).
I will attach the difference between the FAT area before and after the deletion. ([diff_before_and_after_deletion](https://user-images.githubusercontent.com/49133563/145911876-a134f0b7-f717-4322-bc1d-b7a56d431d5b.png))

#### Stacktrace of Excpetion
When I try to read a corrupted file, I get the following exception:
```
IOException
free entry in chain at: 257

org.jnode.fs.jfat.FatChain$ChainIteratåor in next at line 717
org.jnode.fs.jfat.FatChain in read at liåne 362
org.jnode.fs.jfat.FatFile in read at line 92
com.github.magnusja.libaums.javafs.wrapper.fs.UsbFileWrapper in read at line 169
com.github.mjdev.libaums.fs.UsbFileInputStream in read at line 79
kotlin.io.ByteStreamsKt in copyTo at line 110
kotlin.io.ByteStreamsKt in copyTo$default at line 103
```

### Code where problem occurs
I suspect that this can be solved by adding the following code in ByteBlockDevice#write: before arraycopy on line 97 of ByteBlockDevice.kt.
```
targetBlockDevice.read(devOffset, buffer)
buffer.clear()
```
[diff_before_and_after_deletion](https://user-images.githubusercontent.com/49133563/145911876-a134f0b7-f717-4322-bc1d-b7a56d431d5b.png)
[before_deleting_1st_file.img.aa.zip](https://github.com/magnusja/libaums/files/7707888/before_deleting_1st_file.img.aa.zip)
[before_deleting_1st_file.img.ab.zip](https://github.com/magnusja/libaums/files/7707890/before_deleting_1st_file.img.ab.zip)
[before_deleting_1st_file.img.ac.zip](https://github.com/magnusja/libaums/files/7707913/before_deleting_1st_file.img.ac.zip)
[before_deleting_1st_file.img.ad.zip](https://github.com/magnusja/libaums/files/7707918/before_deleting_1st_file.img.ad.zip)
[before_deleting_1st_file.img.ae.zip](https://github.com/magnusja/libaums/files/7707919/before_deleting_1st_file.img.ae.zip)
[before_deleting_1st_file.img.af.zip](https://github.com/magnusja/libaums/files/7707926/before_deleting_1st_file.img.af.zip)
[before_deleting_1st_file.img.ag.zip](https://github.com/magnusja/libaums/files/7707931/before_deleting_1st_file.img.ag.zip)
[before_deleting_1st_file.img.ah.zip](https://github.com/magnusja/libaums/files/7707933/before_deleting_1st_file.img.ah.zip)
[before_deleting_1st_file.img.ai.zip](https://github.com/magnusja/libaums/files/7707934/before_deleting_1st_file.img.ai.zip)
[before_deleting_1st_file.img.aj.zip](https://github.com/magnusja/libaums/files/7707936/before_deleting_1st_file.img.aj.zip)
[before_deleting_1st_file.img.aj.zip](https://github.com/magnusja/libaums/files/7707937/before_deleting_1st_file.img.aj.zip)
[before_deleting_1st_file.img.al.zip](https://github.com/magnusja/libaums/files/7707938/before_deleting_1st_file.img.al.zip)
[before_deleting_1st_file.img.am.zip](https://github.com/magnusja/libaums/files/7707940/before_deleting_1st_file.img.am.zip)
[after_deleting_1st_file.img.aa.zip](https://github.com/magnusja/libaums/files/7707942/after_deleting_1st_file.img.aa.zip)
[after_deleting_1st_file.img.ab.zip](https://github.com/magnusja/libaums/files/7707944/after_deleting_1st_file.img.ab.zip)
[after_deleting_1st_file.img.ac.zip](https://github.com/magnusja/libaums/files/7707946/after_deleting_1st_file.img.ac.zip)
[after_deleting_1st_file.img.ad.zip](https://github.com/magnusja/libaums/files/7707948/after_deleting_1st_file.img.ad.zip)
[after_deleting_1st_file.img.ae.zip](https://github.com/magnusja/libaums/files/7707950/after_deleting_1st_file.img.ae.zip)
[after_deleting_1st_file.img.af.zip](https://github.com/magnusja/libaums/files/7707951/after_deleting_1st_file.img.af.zip)
[after_deleting_1st_file.img.ag.zip](https://github.com/magnusja/libaums/files/7707954/after_deleting_1st_file.img.ag.zip)
[after_deleting_1st_file.img.ah.zip](https://github.com/magnusja/libaums/files/7707955/after_deleting_1st_file.img.ah.zip)
[after_deleting_1st_file.img.ai.zip](https://github.com/magnusja/libaums/files/7707959/after_deleting_1st_file.img.ai.zip)
[after_deleting_1st_file.img.aj.zip](https://github.com/magnusja/libaums/files/7707960/after_deleting_1st_file.img.aj.zip)
[after_deleting_1st_file.img.ak.zip](https://github.com/magnusja/libaums/files/7707961/after_deleting_1st_file.img.ak.zip)
[after_deleting_1st_file.img.al.zip](https://github.com/magnusja/libaums/files/7707962/after_deleting_1st_file.img.al.zip)
[after_deleting_1st_file.img.am.zip](https://github.com/magnusja/libaums/files/7707963/after_deleting_1st_file.img.am.zip)

meguro-pgv, 2022-01-11:
Dear Mr. Magnus @magnusja 
Please let us know if you have any other information you need.
It would be greatly appreciated if you could update libaums in the mavenCentral repository.

magnusja, 2023-03-11:
Should be fixed in v0.9.4

closed 2023-03-11 by magnusja