magnusja/libaums #328: close problem

lawas2008, 2021-12-27:
### Problem
I have a problem,UsbDevice.close(), App crashed.  error log like this:
[image]
 My Code like this:
[image]

[image]

[image]

[image]

magnusja, 2021-12-27:
This happens if you close the USB device twice.

lawas2008, 2021-12-28:
thank you,I am trying...

closed 2021-12-28 by lawas2008