magnusja/libaums #339: Delete empty file causes OutOfMemory issue

AdiSegev, 2022-03-16:
Hi,

I really like your library and found it very useful.

I'm trying to delete an empty file on Fat32 Mass storage.

I'm getting an OutOfMemory Exception:
OutOfMemoryError: Failed to allocate a 124606360 byte allocation with 16773144 free bytes and 109MB until OOM.

I've seen that you've tried to fix it here: https://github.com/magnusja/libaums/commit/36a3d3b3c43eeae6ae8e3f64845200cf3b55c4bb.

In my case I can see that the startCluster is not 0 so you're trying to read untill you reach FAT32_EOF_CLUSTER, but it never happens.

That's my stack trace:

2022-03-16 09:32:23.488 8027-8152/il.co.isrcorp.usbbrowser W/art: Throwing OutOfMemoryError "Failed to allocate a 124606360 byte allocation with 16773144 free bytes and 109MB until OOM"
2022-03-16 09:32:23.495 8027-8152/il.co.isrcorp.usbbrowser E/AndroidRuntime: FATAL EXCEPTION: Thread-2
    Process: il.co.isrcorp.usbbrowser, PID: 8027
    java.lang.OutOfMemoryError: Failed to allocate a 124606360 byte allocation with 16773144 free bytes and 109MB until OOM
        at java.util.Arrays.copyOf(Arrays.java:3231)
        at java.util.Arrays.copyOf(Arrays.java:3204)
        at java.util.ArrayList.grow(ArrayList.java:249)
        at java.util.ArrayList.ensureExplicitCapacity(ArrayList.java:223)
        at java.util.ArrayList.ensureCapacityInternal(ArrayList.java:215)
        at java.util.ArrayList.add(ArrayList.java:441)
        at me.jahnen.libaums.core.fs.fat32.FAT.getChain$libaums_release(FAT.kt:121)
        at me.jahnen.libaums.core.fs.fat32.ClusterChain.<init>(ClusterChain.kt:105)
        at me.jahnen.libaums.core.fs.fat32.FatFile.initChain(FatFile.kt:74)
        at me.jahnen.libaums.core.fs.fat32.FatFile.delete(FatFile.kt:150)
        at il.co.isrcorp.usbbrowser.MainActivity.lambda$delete$1$il-co-isrcorp-usbbrowser-MainActivity(MainActivity.java:292)

Thanks

magnusja, 2022-03-16:
So the file is empty but has a chain? Not sure if that is correct ...

Related to https://github.com/magnusja/libaums/issues/112 maybe?

AdiSegev, 2022-03-19:
Hi,

Yes I think it has a chain.

Any suggestions how can it be solved?

muskokatech, 2022-05-06:
I am experiencing an identical problem. Infinite loop on delete. In my  case the file is very small (about 100 bytes).

magnusja, 2023-03-11:
Fixed with v0.9.4

closed 2023-03-11 by magnusja