magnusja/libaums #346: ZipOutputStream from UsbFile probem

talhacomak, 2022-06-25:
### Problem
I'm trying to back up the data of a program on a smartphone to a usb memory stick via OTG. When i create a ZipOutputStream from an OutputStream of a UsbFile object, ZipOutputStream functions like putNextEntry or write doesn't work without of an error or warning. However, when i use my code for internal storage File, everything is working well. Also when i try to create a file or write something into external storage by UsbFile.write() function or Outputstream.write(bytes) function, everything is working well. The only problem is creating ZipOutputStream object from UsbFile object. 

### Expected behavior

UsbFile dstFile, FileSystem currentFs
```
OutputStream os = UsbFileStreamFactory.createBufferedOutputStream(dstFile, currentFs);
//OutputStream os = new UsbFileOutputStream(dstFile);
ZipOutputStream zipOut = new ZipOutputStream(os);
```

I am expecting a ZipOutputStream object that can perform putNextEntry, write functions without a problem. 

### Actual behavior

ZipOutputStream object that can't perform putNextEntry, write functions There is no error or warning. And the output zip file is broken.

### Code where problem occurs
```
try{
    ZipEntry zipEntry = new ZipEntry(thPath + "/");
    zipOut.putNextEntry(zipEntry);
    zipOut.closeEntry();
}
catch (Exception e){
    e.printStackTrace();
}
```
Actually there is no error or warning. Just it produced broken zip file.

magnusja, 2022-11-22:
Does `closeEntry` call `close()` of the decorated output stream? Generally I would advice to call close on the file system structure.

talhacomak, 2022-11-22:
`closeEntry()` doesn't call `close()`. Is it required to call close too?

magnusja, 2022-11-22:
It is actually quite important to call close.

https://github.com/magnusja/libaums/blob/904fe83fce727d8a6ba241fe8ee9ee946b2a80da/libaums/src/main/java/me/jahnen/libaums/core/fs/UsbFileOutputStream.kt#L57

https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/me/jahnen/libaums/core/fs/fat32/FatFile.kt#L117-L130

As this is writing the correct file length, I could imagine that this is actually quite important. Otherwise even when reading all the clusters of the chain, you dont know where to stop in the last cluster which easily could corrupt a zip file.

magnusja, 2023-03-11:
Is this still happening with v0.9.4? There has been a nasty bug fixed in https://github.com/magnusja/libaums/pull/374

closed 2023-03-20 by magnusja