magnusja/libaums #348: Folder copy ??

mathan1079, 2022-07-02:
How to copy folder on USB to internal directory . Please give me the example

naveenrobo, 2022-07-07:
You could you the below to copy folders and contents inside 😁

```kt
currentFs.rootDirectory.listFiles().forEach {
                    if (it.name == "Folder name") {
                        val list = getListUsbFiles(it)
                        try {
                            list.forEach { rec ->
                                val dw = File("${filesDir}${rec.absolutePath}")
                                if (!File(dw.parent).exists()) {
                                    File(dw.parent).mkdirs()
                                }

                                copyFile(
                                    currentFs,
                                    rec,
                                    dw.absolutePath
                                )
                            }
                        } catch (e: Exception) {
                            e.printStackTrace()
                        }
                    }
                }
```

```kt
fun getListUsbFiles(parentDir: UsbFile): List<UsbFile> {
    val inFiles: ArrayList<UsbFile> = ArrayList()
    val files: Array<UsbFile> = parentDir.listFiles()
    for (file in files) {
        if (file.isDirectory) {
            inFiles.addAll(getListUsbFiles(file))
        } else {
            inFiles.add(file)
        }
    }
    return inFiles
}

fun copyFile(currentFs: FileSystem, input: UsbFile, output: String) {
    val inputStream: InputStream = createBufferedInputStream(input, currentFs)

    File(output).copyInputStreamToFile(inputStream)
}

fun File.copyInputStreamToFile(inputStream: InputStream) {
    this.outputStream().use { fileOut ->
        inputStream.copyTo(fileOut)
    }
}
```

magnusja, 2022-11-22:
This is a feature of the example application even?!

closed 2022-11-22 by magnusja