magnusja/libaums #350: PdfDocument writeTo UsbFileOutputStream, File corruption

jie65535, 2022-08-02:
### Background
I am using `libaums` to export `pdf` to `u-disk`.

### Problem
Following the example, I create `UsbFileOutputStream/BufferedOutputStream` and then use `pdfDocument.writeTo(stream)` to write the pdf to `u-disk`.

On the PC, I cannot open this file because it is corrupted.

---

- Using the same method, write the pdf to the `sdcard`, get the file through `adb pull`, normal.
- Write the pdf to the memory cache (ByteArray), then `copyTo` to `u-disk`, the file is normal.
- Write pdf directly to `u-disk`, file corrupted.

### Code where problem occurs
```kotlin

val cache = ByteArrayOutputStream().use {
    exportPdf(it)
    it.toByteArray()
}

//  val cache = File(Environment.getExternalStoragePublicDirectory(Environment.DIRECTORY_DOWNLOADS), filename)
//  cache.outputStream().use {
//   exportPdf(it)
// }

val file = fs.rootDirectory.createFile(filename)
UsbFileStreamFactory.createBufferedOutputStream(file, fs).use { outStream ->
    cache.inputStream().use { inStream ->
        inStream.copyTo(outStream, fs.chunkSize)
    }
// exportPdf(outStream)  // file corrupted.
}

private fun exportPdf(os: OutputStream) {
  val pdfDocument = PdfDocument()
  try {
      val pageInfo = PdfDocument.PageInfo.Builder(595, 842, 1).create()
      val page = pdfDocument.startPage(pageInfo)
      drawPage(page)
      pdfDocument.finishPage(page)
      pdfDocument.writeTo(os)
      fs.flush()
  } finally {
      pdfDocument.close()
  }
}
```

---

I solved the problem temporarily using memcache for now, but I'm wondering why this is happening as I've wasted a whole day with this problem.

### More info

The corrupted file is the exact same size as the normal file, but by comparing the files, most of the content is the same, with only a few differences.

[image]

[image]

_(I wrote some code to convert pdf binary to hex string text for VSCode comparison)_

jie65535, 2022-08-02:
This seems to be the same as #346

magnusja, 2022-11-22:
So to summarize, if you use the the usb outputstream and feed it into the pdf library it breaks. But if you write to somewhere else first and then copy the pdf file it is working?

Makes me wonder if there are some methods in the outputstream which are not working properly?

magnusja, 2023-03-11:
Is this still happening with v0.9.4? There has been a nasty bug fixed in https://github.com/magnusja/libaums/pull/374

jie65535, 2023-03-12:
> Is this still happening with v0.9.4? There has been a nasty bug fixed in #374

I will try v0.9.4. My code is still reserved, but it may take a few days. Thank you for your work!

jie65535, 2023-03-15:
The test is completed. v0.9.4 has fixed this problem. Thank you for your work!

closed 2023-03-15 by jie65535