magnusja/libaums #357: Copy file Time stamp issue

ssdulawat, 2022-11-02:
### Problem
When copy file from internal storage to USB storage that time the file copy with current time stamp

### Expected behavior
Copy file with same time stamp from source file

### Actual behavior

#### Stacktrace of Excpetion

```
(if exists)
```

### Code where problem occurs
  
      InputStream inputStream = new BufferedInputStream(App.getContext().getContentResolver().openInputStream(uri));
      OutputStream outputStream = UsbFileStreamFactory.createBufferedOutputStream(file, currentFs);
      try {
        byte[] bytes = new byte[currentFs.getChunkSize()];
        int count;
        long bytesWritten = 0;

        while ((count = inputStream.read(bytes)) != -1) {

          if (stopWrite) {
            Timber.d("Copying interrupted!");
            file.delete();
            break;
          }

          outputStream.write(bytes, 0, count);

          bytesWritten += count;
          int progress = (int) (((double) bytesWritten / (double) size) * 100.0);
          if (progressListener != null) progressListener.onProgress(progress, 100);
        }

        outputStream.close();
        inputStream.close();

ssdulawat, 2022-11-02:
Please let me know for the solution.
I am using [libaums 0.7.4 ](libaums:0.7.4)

magnusja, 2022-11-22:
You are copying a stream of bytes, you have to set the timestamps yourself. This is currently not exposed over the public API but probably not too hard to add.

ssdulawat, 2023-04-03:
How to set timestamp on outputStream  file that library method there is no way to add it. please suggest me any example.

magnusja, 2023-04-03:
Currently, there is no way. You would need to add this to the `UsbFile` API. I am happy to take a Pull Request for that :)