magnusja/libaums #384: crash when calling init method of UsbMassStorageDevice.

nondeepshit, 2023-03-20:
### Problem
crash when setup device

### Expected behavior
succed to get the root directory 

### Actual behavior
crash when calling init method of UsbMassStorageDevice.

#### Stacktrace of Excpetion

```
FATAL EXCEPTION: main
Process: com.aispeech.dui.dds.usbotgdemo, PID: 1252
java.lang.RuntimeException: Error receiving broadcast Intent { act=com.newsmy.farosh.USB_PERMISSION flg=0x10 hwFlg=0x100 (has extras) } in com.aispeech.dui.dds.usbotgdemo.MainActivity$1@e688b84
	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1796)
	at android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run(Unknown Source:2)
	at android.os.Handler.handleCallback(Handler.java:900)
	at android.os.Handler.dispatchMessage(Handler.java:103)
	at android.os.Looper.loop(Looper.java:219)
	at android.app.ActivityThread.main(ActivityThread.java:8668)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1109)
Caused by: java.lang.IllegalArgumentException
	at java.nio.Buffer.limit(Buffer.java:290)
	at java.nio.ByteBuffer.limit(ByteBuffer.java:818)
	at me.jahnen.libaums.core.driver.scsi.ScsiBlockDevice.transferOneCommand(ScsiBlockDevice.kt:270)
	at me.jahnen.libaums.core.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:154)
	at me.jahnen.libaums.core.driver.scsi.ScsiBlockDevice.initAttempt(ScsiBlockDevice.kt:105)
	at me.jahnen.libaums.core.driver.scsi.ScsiBlockDevice.init(ScsiBlockDevice.kt:83)
	at me.jahnen.libaums.core.UsbMassStorageDevice.setupDevice(UsbMassStorageDevice.kt:138)
	at me.jahnen.libaums.core.UsbMassStorageDevice.init(UsbMassStorageDevice.kt:100)
	at com.aispeech.dui.dds.usbotgdemo.MainActivity.setupDevice(MainActivity.java:118)
	at com.aispeech.dui.dds.usbotgdemo.MainActivity.access$000(MainActivity.java:25)
	at com.aispeech.dui.dds.usbotgdemo.MainActivity$1.onReceive(MainActivity.java:42)
	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1786)
	at android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run(Unknown Source:2) 
	at android.os.Handler.handleCallback(Handler.java:900) 
	at android.os.Handler.dispatchMessage(Handler.java:103) 
	at android.os.Looper.loop(Looper.java:219) 
	at android.app.ActivityThread.main(ActivityThread.java:8668) 
	at java.lang.reflect.Method.invoke(Native Method) 
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513) 
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1109) 
```

### Code where problem occurs
`private void discoverDevice(Intent intent) {
	usbManager = (UsbManager) getSystemService(Context.USB_SERVICE);
	massStorageDevices = UsbMassStorageDevice.getMassStorageDevices(this);
	if (massStorageDevices.length == 0){
		return;
	}
	usbDevice = intent.getParcelableExtra(UsbManager.EXTRA_DEVICE);

	Log.i("MainActivity :: " , "usbManager.hasPermission == " + usbManager.hasPermission(usbDevice));

	if (usbDevice != null && usbManager.hasPermission(usbDevice)){
		setupDevice();
	}else {
		PendingIntent pendingIntent = PendingIntent.getBroadcast(this
				, 0, new Intent(ACTION_USB_PERMISSION), 0);
		usbManager.requestPermission(massStorageDevices[0].getUsbDevice(), pendingIntent);
	}
}

private void setupDevice() {
	try {
		Log.e("MainActivity", "massStorageDevices size: " + massStorageDevices.length);
		massStorageDevices[0].init();
		Log.i("MainActivity", "massStorageDevices init success ");
		// we always use the first partition of the device
		FileSystem currentFs = massStorageDevices[0].getPartitions().get(0).getFileSystem();
		UsbFile root = currentFs.getRootDirectory();
		textView.setText(root.getName());

	} catch (IOException e) {
		Log.i("MainActivity", "setupDevice: exception == "+ e.getMessage());
	}
}`

magnusja, 2023-03-23:
So your device fails when trying to test if the unit is ready. The OP has the problem that the `ScsiInquiry` does not work.

So afaik testing if the unit is ready, has no data phase. But your device seems to respond with some more data.

xzeman43, 2023-03-24:
Then I am sorry for exploiting this thread. However, it is interesting that both devices are standard USB disks formatted to the same filesystem. I will try to dig deeper and find out why they do no behave the same.

xzeman43, 2023-03-27:
> So your device fails when trying to test if the unit is ready. The OP has the problem that the `ScsiInquiry` does not work.
> 
> So afaik testing if the unit is ready, has no data phase. But your device seems to respond with some more data.

Hello, so quick update on the issue you described. For whatever reason I did not see it straight away but it seems that the error occurs if the `device.init() ` happens too fast after the device attachment. When I launch the procedure straight from `UsbManager.ACTION_USB_DEVICE_ATTACHED` callback without any delay, 99/100 attempts fail on the `java.lang.IllegalArgumentException` .However, when I introduce a slight delay (one sec seems stable) after the USB attach event, it works flawlessly. Maybe it will be a pleasant update to have a method that will be checking for this specific issue - it seems like after the USB attach broadcast, there is still some initialization running in the background that messes up with your library.

viktor76525, 2023-05-11:
Hello, I also bumped into this issue. The workaround is to wait 900+ms before calling init.
It only happens on certain USB drives, not sandisk, mostly older usb 2 and no-name ones.
It was ok on 0.9.1, couldn't get intermediate versions to build due to some dependency issues, and then it is in both 0.9.4 and 0.10.0.

xzeman43, 2023-05-11:
> Hello, I also bumped into this regression. The workaround is to wait 900+ms before calling init. It only happens on certain USB drives, not sandisk, mostly older usb 2 and no-name ones. It was ok on 0.9.1, couldn't get intermediate versions to build due to some dependency issues, and then it is in both 0.9.4 and 0.10.0.

Yes, that seems in line with my observations. But still, from time to time it breaks the connection with the device and it has to be reconnected. So the solution was quick and easy to implement, but somehow its not 100% bulletproof.