magnusja/libaums #385: hey, Copy the meeting report through fat32 here  Error transferring command; errno 0 null

zhangpeng1112, 2023-03-24:
UsbFile targetFile = null;
        try {
            targetFile = usbFile.createFile(fileName);
            FileInputStream fis = new FileInputStream(file);
            //int size = fis.available();
            long size = file.length();
            UsbFileOutputStream uos = new UsbFileOutputStream(targetFile);
            //8K 缓冲区
            byte[] buffer = new byte[1024 * 16];
            //byte[] buffer = new byte[(int) Constants.BytesPerCluster];
            long readPos = 0;
            long writePos = 0;
            LogUtils.v("PtoU copy " + fileName + " to " + targetFile.getAbsolutePath());
            long time1 = System.currentTimeMillis();
            while ((readPos = fis.read(buffer)) != -1) {
                uos.write(buffer, 0, (int) readPos);
                writePos += readPos;
                if (listener != null) {
                    listener.onProgress((int) (writePos * 100l / size));
                    LogUtils.v("PtoU write file with progress = " + (int) (writePos * 100l / size));
                }
            }
            uos.flush();
            fis.close();
            uos.close();

magnusja, 2023-03-24:
Try `libusbcommunication` https://github.com/magnusja/libaums/tree/develop/libusbcommunication

closed 2023-03-24 by magnusja

zhangpeng1112, 2023-03-27:
Sorry, this libusbcommunication may also have errors
jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):86 libusb_bulk_transfer returned -7, Operation timed out

zhangpeng1112, 2023-03-27:
device:aigo Device