magnusja/libaums #386: Do I need to integrate the libusbcommunication library to read devices in the ntfs system

wangshufu, 2023-03-24:
devices.forEachIndexed { index, usbMassStorageDevice ->
                    usbMassStorageDevice.init()
                    // Only uses the first partition on the device
                    LogUtils.d("***", "index: $index ${usbMassStorageDevice.partitions?.size}")
                    if (usbMassStorageDevice.partitions.isNotEmpty()) {
                        val currentFs = usbMassStorageDevice.partitions[0].fileSystem
                        Log.d("***", "Capacity: " + currentFs.capacity)
                        Log.d("***", "Occupied Space: " + currentFs.occupiedSpace)
                        Log.d("***", "Free Space: " + currentFs.freeSpace)
                        Log.d("***", "Chunk size: " + currentFs.chunkSize)
                    }
 }

Occasionally, data is returned!
Here is my MainActivity code...

class MainActivity : AppCompatActivity(),ConsoleUtils.ConsoleMessageListener {

    companion object {
        init {
            registerFileSystem(JavaFsFileSystemCreator())
        }
    }

I have a question. I want to read the USB on the NTFS system. Do I need to add this code?

registerCommunication(LibusbCommunicationCreator())
underlyingUsbCommunication = UsbCommunicationFactory.UnderlyingUsbCommunication.OTHER

wangshufu, 2023-03-24:
in addition:
An exception occurred while executing this line of code...
usbMassStorageDevice.init()

java.lang.IllegalArgumentExceptionkotlin.Unit

After multiple calls, throw
java.io.IOException: MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try againkotlin.Unit

magnusja, 2023-03-24:
No the `libusbcommunication` module has nothing to do with the file system.
 You will need to use `java-fs`. Note that it only has read only support for ntfs though.

closed 2023-03-24 by magnusja

wangshufu, 2023-03-27:
Thank you, I have a question. Can NTFS achieve read and write functions? If possible, what should be done? If it can only be read-only, then why?

wangshufu, 2023-03-30:
Hello, the system is FAT, but it still cannot be written. Do I have to go to the system file manager to authorize it? If so, how do I authorize it?
[image]

magnusja, 2023-03-30:
I dont know what that means (I only speak German and English).

Also I dont recognize that app. Please use the example app.
If you coded that app, I guess it is upon you to enable writing and remove that blue banner?!

wangshufu, 2023-04-07:
Hello, which systems are supported in the current version with read and write functions; I tried the exfat system and found an exception when calling usbMassStorageDevice. init()

java.io.IOException: MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try again