magnusja/libaums #396: root.listFiles.size() is zero when change the disk filie system to exFat

swithun-liu, 2023-08-26:
### Problem

[image]

rootFile.listFiles() size is zero

[image]

it works well before I change the disk file system to exFat for big file

### Expected behavior

### Actual behavior

#### Stacktrace of Excpetion

```
(if exists)
```

### Code where problem occurs

swithun-liu, 2023-08-27:
[image]

I tried to add java-fs

but what should I do , I am really confused

magnusja, 2023-09-06:
See this line here https://github.com/magnusja/libaums/blob/develop/app/src/main/java/me/jahnen/libaums/core/usbfileman/MainActivity.kt#L80

Highly unstable tho!

closed 2023-09-06 by magnusja