magnusja/libaums #400: An error was encountered when calling the UsbMassStorageDevice#init method

zengjie-android, 2023-09-27:
### Problem
I have changed three memory cards that support fat32,but still failed at init method, here is some log maybe help me analyze what the problem is, awaiting your reply, thanks

### Expected behavior
No more errors when calling the UsbMassStorageDevice#init method

### Actual behavior

#### Stacktrace of Excpetion

```
09-27 10:28:25.318 25962 25962 I ScsiBlockDevice: Block size: 512 09-27 10:28:25.318 25962 25962 I ScsiBlockDevice: Last block address: 125173759 09-27 10:28:25.337 25962 25962 D ScsiBlockDevice: inquiry response: ScsiInquiryResponse [peripheralQualifier=0, peripheralDeviceType=0, removableMedia=false, spcVersion=6, responseDataFormat=2] 09-27 10:28:25.339 25962 25962 E ScsiBlockDevice: Error transferring command; errno 0 null 09-27 10:28:25.339 25962 25962 D ScsiBlockDevice: Reset bulk-only mass storage 09-27 10:28:25.340 25962 25962 D ScsiBlockDevice: Trying to clear halt on both endpoints 09-27 10:28:25.842 25962 25962 E ScsiBlockDevice: Error transferring command; errno 0 null 09-27 10:28:25.843 25962 25962 D ScsiBlockDevice: Trying to reset the device 09-27 10:28:26.620 25962 25962 E ScsiBlockDevice: Error transferring command; errno 0 null 09-27 10:28:26.620 25962 25962 D ScsiBlockDevice: Reset bulk-only mass storage 09-27 10:28:26.620 25962 25962 D ScsiBlockDevice: Trying to clear halt on both endpoints```

### Code where problem occurs

zengjie-android, 2023-09-27:
implementation 'com.github.mjdev:libaums:0.7.4'

zengjie-android, 2023-09-27:
The original version is `implementation 'com.github.mjdev:libaums:0.7.4'`

magnusja, 2023-11-22:
For some reason connection to USB is lost and cant recover. Maybe try another otg adapter