magnusja/libaums #406: Why does mass storage send a 16k data packet into multiple data packets?

junxiaojun, 2023-11-23:
### Problem
 private fun transferOneCommand(command: CommandBlockWrapper, inBuffer: ByteBuffer): Int

I have a 16k data packet. Why is it divided into 5 reads? Can I read 16k data packets at once? Instead of subcontracting to read

### Expected behavior

### Actual behavior

#### Stacktrace of Excpetion

```
(if exists)
```

### Code where problem occurs

junxiaojun, 2023-11-23:
@magnusja

magnusja, 2023-11-24:
Depends on the flash drive. Oftentimes, some support larger ones, some do not. 

But anyways, the ScsiWrite and Read commands currently only support some smaller chunk sizes. 
See here, https://github.com/magnusja/libaums/blob/develop/libaum
[image]
s/src/main/java/me/jahnen/libaums/core/driver/scsi/commands/ScsiRead10.kt

Currently, there is a read command 32, which probably speeds up transfer by a lot if usb drive supports it. See here: https://www.seagate.com/files/staticfiles/support/docs/manual/Interface%20manuals/100293068j.pdf
[image]

I am happy to take a PR. But as the USB spec is complex, it is hard to support each of these flawlessly.

junxiaojun, 2023-12-08:
@magnusja 
Thank you. I am very happy to see your reply. The problem has been solved. Thank you very much.