magnusja/libaums #409: Cannot read a UsbFile content

auloma04, 2023-12-11:
Hello everyone. I am trying to read the content of my USB device.  

I can successfully list the content of the USB key, list files etc ..  
I am supposed to have a file named "network.ini" on the USB key. The file is well listed

### Problem
When I try to read the content of the file, i get the error:  *java.io.IOException: MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try again* 
After having looked at the logcat, I saw that the Stack trace of the error mentions the error : *System.err: Caused by: java.io.IOException: Could not read from device, result == -1 errno 0 null*  
  
So following the documentation advices, I tried to use libusb:  
- Adding the right implementation in gradle dependencies
- Adding at the begining of my MainActivity:
```
UsbCommunicationFactory.registerCommunication(new LibusbCommunicationCreator());
UsbCommunicationFactory.setUnderlyingUsbCommunication(UsbCommunicationFactory.UnderlyingUsbCommunication.OTHER);
```
An still get an error when try to read the InputStream: *java.io.IOException: MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try again*  
But this time logcat is not pointing the same one on the stack trace: *Caused by: me.jahnen.libaums.libusbcommunication.LibusbException: libusb control transfer failed: Input/output error [-1]*

### Expected behavior
What I want is to convert the content of the file into a String. (its content is very light)  

How I do that after having accessed the fileSystem of the UsbMassStorageDevice:
```java
public void readFs(FileSystem fs) throws IOException {
        Log.d(TAG, "Capacity: " + fs.getCapacity());
        Log.d(TAG, "Occupied Space: " + fs.getOccupiedSpace());
        Log.d(TAG, "Free Space: " + fs.getFreeSpace());
        Log.d(TAG, "Chunk size: " + fs.getChunkSize());

        UsbFile root = fs.getRootDirectory();

        UsbFile[] files = root.listFiles();
        for(UsbFile file: files) {
            Log.d(TAG, "File:" + file.getName() + " is a dir: " + file.isDirectory());
            if(!file.isDirectory()) {
                readFile(file, fs);
            }
        }
    }

    public void readFile(UsbFile file, FileSystem fs) throws IOException {
        Log.d(TAG, "readFile: " + file.getName());
        InputStream is = UsbFileStreamFactory.createBufferedInputStream(file, fs);
        byte[] buffer = new byte[fs.getChunkSize()];

        is.read(buffer);
    }
```
I spent a lot of time on this issue and have no more ideas to solve this quite urgent problem ...
Thanks for your help

auloma04, 2023-12-12:
Github for reproduction: https://github.com/auloma04/android-usb-reader

magnusja, 2024-01-25:
* Try not using `libusb` but the Android communication, does that work?
* Try a different OTG adapter. Sometimes these are not the best for stable communication.

Unfortunately, it looks like a low level communication errror. So not much I can help. Do other apps allowing USB access work?

hsycc, 2024-06-07:
Hello, have you solved this problem yet @auloma04