magnusja/libaums #411: [Enhancement] Add infrastructure for testing the low-level SCSI driver

depau, 2023-12-24:
As mentioned in #410 I think it could be meaningful to add infrastructure for testing low-level SCSI functionality. It is generally robust, though we both very well know how easy it is to mess up `ByteBuffer` usage and forget a `.flip()` here or a `.clear()` there.

I was thinking we could add a `UsbCommunicationRecorder` that proxies a real `UsbCommunication` and records all commands to a `.pcapng` or similar, as well as a `UsbCommunicationPlayer` that reads the capture, asserts that sent commands match the output and replays responses.

Then we can have a shared set of test cases that can be called from instrumented tests to generate the capture, and from regular unit tests to ensure the commands are correct.

What do you think?

magnusja, 2024-01-12:
Hey @depau 

thanks a lot for your message. Generally, yes. This sounds like a great idea!

I have to be honest though. My time but also my interests shifted in the last years. So I am unable to put a resonable amount of work into this. You already have collaboration access to this repository. Not sure how to make you an admin though. Maybe I have to transfer to an organization.

But I would give you full responsiblity. I obviously would be happy to review code. But I think in the mid term future I wont write any code myself for this.

Thanks for your continued contribution. It is really awesome to have at least someone helping out. Mostly people are just requesting bug fixes or new features instead of contributing. (although I guess over the time there have been several one time contributions which is also great. But I think this whole topic is also so complex to get 100% right it is alsmost impossible for an individual to pull off on the side)

depau, 2024-01-12:
Hey!

> I have to be honest though. My time but also my interests shifted in the last years. So I am unable to put a resonable amount of work into this. You already have collaboration access to this repository. Not sure how to make you an admin though. Maybe I have to transfer to an organization.

I absolutely understand.

> Thanks for your continued contribution. It is really awesome to have at least someone helping out. Mostly people are just requesting bug fixes or new features instead of contributing. (although I [...]

You're welcome and thank you too for the reviews and for the quality of the code! I'm happy to hear that you do occasionally get contributions

I fully understand your position about the project and I feel the same for many of my projects. Speaking about EtchDroid specifically, after lots of frustration and burn-out I decided I want to keep it somewhat alive, dedicating some free time to it when I can and feel like it. This implies I'm definitely interested in keeping libaums alive, the core functionality in particular. Since real-life devices have their own quirks, this type of functionality is very valuable.

Since you like the idea I will work on it over the coming months (or maybe years, let's see how my free time, interests and stress fluctuate).

> But I would give you full responsiblity. I obviously would be happy to review code. But I think in the mid term future I wont write any code myself for this.

I think I do have write access to the repo, if I need help I'll ping you when I start working on it as well as when it's ready for reviewing.

Thanks for your reply!