magnusja/libaums #414: Play Music From USB Pendrive

BhagavathNath, 2024-03-04:
Hi let me clear a doubt first , is this the only library to read files from the USB devices. Can you suggest any libraries is there any way android provide to read files from USB other than open source libraries

BhagavathNath, 2024-03-04:
I am a completer beginner to android so I don't know much more about the libraries and all. I searched a lot and I found this as the only solution to read from the USB. Please help me to verify

Ace122, 2024-03-27:
You can use Storage Access Framework(SAF)

Ace122, 2024-03-27:
> Hi let me clear a doubt first , is this the only library to read files from the USB devices. Can you suggest any libraries is there any way android provide to read files from USB other than open source libraries

You can use Storage Access Framework(SAF)