magnusja/libaums #416: Could not read from device, result == -1 errno 0 null

motosheep, 2024-03-16:
### Problem
runing program in the redmi mobile phone with android13 system。when use the api "createdirectory(String xxx)"，the app no response and tips anr error after few second.

### Expected behavior
can create the usb diretory

### Actual behavior

#### Stacktrace of Excpetion

2024-03-16 21:08:41.093 10315-10315 ScsiBlockDevice         com.example.demo                     W  Could not read from device, result == -1 errno 0 null, retrying...
2024-03-16 21:08:41.194 10315-10315 System.err              com.example.demo                     W  java.io.IOException: MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try again
2024-03-16 21:08:41.195 10315-10315 System.err              com.example.demo                     W  	at me.jahnen.libaums.core.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:190)
2024-03-16 21:08:41.195 10315-10315 System.err              com.example.demo                     W  	at me.jahnen.libaums.core.driver.scsi.ScsiBlockDevice.initAttempt(ScsiBlockDevice.kt:105)
2024-03-16 21:08:41.196 10315-10315 System.err              com.example.demo                     W  	at me.jahnen.libaums.core.driver.scsi.ScsiBlockDevice.init(ScsiBlockDevice.kt:83)
2024-03-16 21:08:41.196 10315-10315 System.err              com.example.demo                     W  	at me.jahnen.libaums.core.UsbMassStorageDevice.setupDevice(UsbMassStorageDevice.kt:138)
2024-03-16 21:08:41.196 10315-10315 System.err              com.example.demo                     W  	at me.jahnen.libaums.core.UsbMassStorageDevice.init(UsbMassStorageDevice.kt:100)
(if exists)
```

### Code where problem occurs
           fileSystem.createDirectory("dd");

PratikLahagora, 2024-06-14:
Hello @motosheep Have you found any solution ?

magicbaby810, 2025-12-24:
`Could not read from device, result == -1 errno 0 null
com.xxx.utils.CommonUtils.xxx(CommonUtils.kt:905)
出错堆栈

1	java.io.IOException:MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try again
2	me.jahnen.libaums.core.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:190)
3	......
4	Caused by:
5	java.io.IOException:Could not read from device, result == -1 errno 0 null
6	me.jahnen.libaums.core.usb.JellyBeanMr2Communication.bulkOutTransfer(JellyBeanMr2Communication.kt:41)
7	me.jahnen.libaums.core.driver.scsi.ScsiBlockDevice.transferOneCommand(ScsiBlockDevice.kt:264)
8	me.jahnen.libaums.core.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:154)
9	me.jahnen.libaums.core.driver.scsi.ScsiBlockDevice.read(ScsiBlockDevice.kt:331)
10	me.jahnen.libaums.core.driver.ByteBlockDevice.read(ByteBlockDevice.kt:54)
11	me.jahnen.libaums.core.fs.fat32.FAT.getChain$libaums_release(FAT.kt:128)
12	me.jahnen.libaums.core.fs.fat32.ClusterChain.<init>(ClusterChain.kt:105)
13	me.jahnen.libaums.core.fs.fat32.FatFile.initChain(FatFile.kt:74)
14	me.jahnen.libaums.core.fs.fat32.FatFile.read(FatFile.kt:101)
15	me.jahnen.libaums.core.fs.UsbFileInputStream.read(UsbFileInputStream.kt:79)

@PratikLahagora  

 I fixed it just now.  It took me two days .  The "MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try again", it's so confusing. The truth is “usbfs: process 3402 (DefaultDispatch) did not claim interface 0 before use”. It's exactly the source of the problem.  I performed the "usbMassStorageDevice?.init()" twice in my code.