magnusja/libaums #418: 为什么读取NTFS、exFAT文件系统的SD卡数据很慢？读取FAT32就很快？

Kotlin2022, 2024-04-26:
我用了不同文件系统的SD卡测试读取SD卡上的照片，发现NTFS、exFAT文件系统的SD卡数据很慢，读取FAT32就很快，
代码如下：
` 
    private fun setupDevice() {
        try {
            massStorageDevices[currentDevice].init()

            // we always use the first partition of the device
            currentFs = massStorageDevices[currentDevice].partitions[0].fileSystem.also {
                Log.d(TAG, "Capacity: " + it.capacity)
                Log.d(TAG, "Occupied Space: " + it.occupiedSpace)
                Log.d(TAG, "Free Space: " + it.freeSpace)
                Log.d(TAG, "Chunk size: " + it.chunkSize)
            }

            val root = currentFs.rootDirectory
            val images =   searchImages(root.listFiles())
            images?.apply {
                Log.d(TAG,"images.size = $size") 
            }
        } catch (e: IOException) {
            Log.e(TAG, "error setting up device", e)
        }
    }
@Throws(IOException::class)
    fun searchImages(usbFiles: Array<UsbFile>): ArrayList<UsbFile>? {
        val files = ArrayList<UsbFile>()
        for (usbFile in usbFiles) {
            if (usbFile.isDirectory && usbFile.name == "DCIM") { //找到相册
                val photosFolder = usbFile.listFiles()
                Log.d(TAG, photosFolder.contentToString())
                if (photosFolder.isNotEmpty()) { //相册下的文件夹大于0
                    for (file in photosFolder) {
                        if (file.isDirectory) {
                            val images = file.listFiles()
                            Log.d(TAG, images.contentToString())
                            for (image in images) {
                                if (image.name.endsWith("JPEG") || image.name.endsWith("JPG")) {
                                    files.add(image)
                                }
                            }
                        }
                    }
                }
            }
        }
        return files
    }`
同样1500张图片，NTFS、exFAT文件系统的SD卡需要1分多钟才能读完，FAT32文件系统的SD卡只需要几秒钟，这时为什么？

Kotlin2022, 2024-04-26:
我发现实际上是usbFile.listFiles() 这个方法很慢，不清楚为什么？

Kotlin2022, 2024-04-26:
`    @Throws(IOException::class)
    fun refresh() {
        dialog.setMessage("loading...")
        dialog.show()
        CoroutineScope(Dispatchers.IO).launch {
            Log.d(MainActivity.TAG,"开是检索文件")
            val usbFiles = currentDir.listFiles()
            files = listOf(*usbFiles)
            Log.d(MainActivity.TAG,"文件检索完成 files.size = ${files.size} ${usbFiles.contentToString()}")
            Collections.sort(files, comparator)
            withContext(Dispatchers.Main) {
                notifyDataSetChanged()
                dialog.dismiss()
            }
        }

    }`

15:56:05.770 25808-27815 USB_lib    com...usbfileman  D  开是检索文件
15:56:44.140 25808-27815 USB_lib    com...usbfileman  D  文件检索完成 files.size = 1 [100MSDCF]

Kotlin2022, 2024-04-26:
39秒才执行完 currentDir.listFiles()  是不是太慢了？

zhishenglin, 2024-05-23:
请问下你是怎么读取ntfs文件系统的，我可以读取fat32的u盘。但是当我读取ntfs的移动硬盘时，在执行到init()初始化设备方法时就抛出了不支持的文件系统异常

SsmallLucky1, 2025-02-08:
> 请问下你是怎么读取ntfs文件系统的，我可以读取fat32的u盘。但是当我读取ntfs的移动硬盘时，在执行到init()初始化设备方法时就抛出了不支持的文件系统异常

有些安卓版本系统，不支持NFTS文件系统，要装驱动。