magnusja/libaums #423: Could not create file and directory on usb storage formated by exfat while running app

grantfee, 2024-06-16:
### Problem
Could not create file and directory on usb storage formatted by exFat while running app， on android 11 and 13 platform

### Expected behavior

### Actual behavior

#### Stacktrace of Excpetion

```
UsbMassStorageDevice     I  found usb device: /dev/bus/usb/002/002=UsbDevice[mName=/dev/bus/usb/002/002,mVendorId=2316,mProductId=8992,mClass=0,mSubclass=0,mProtocol=0,mManufacturerName=KOWIN,mProductName=KOWIN Portable SSD,mVersion=1.00,mSerialNumberReader=android.hardware.usb.IUsbSerialReader$Stub$Proxy@b602b78, mHasAudioPlayback=false, mHasAudioCapture=false, mHasMidi=false, mHasVideoCapture=false, mHasVideoPlayback=false, mConfigurations=[
                                                                UsbConfiguration[mId=1,mName=null,mAttributes=128,mMaxPower=250,mInterfaces=[
                                                                UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=8,mSubclass=6,mProtocol=80,mEndpoints=[
                                                                UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0]
                                                                UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]]]]
2024-06-16 12:06:01.963 28088-28088 UsbMassStorageDevice     I  Found usb interface: UsbInterface[mId=0,mAlternateSetting=0,mName=null,mClass=8,mSubclass=6,mProtocol=80,mEndpoints=[
                                                                UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0]
                                                                UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
2024-06-16 12:06:01.963 28088-28088 UsbMassStorageDevice     I  Found usb endpoint: UsbEndpoint[mAddress=1,mAttributes=2,mMaxPacketSize=512,mInterval=0]
2024-06-16 12:06:01.963 28088-28088 UsbMassStorageDevice     I  Found usb endpoint: UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]
2024-06-16 12:06:01.964 28088-28088 MainActivity             D  received usb device via intent
2024-06-16 12:06:01.966 28088-28088 native_libusbcom         D  jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeInit(JNIEnv *, jobject, jint, jlongArray):11 init native libusb
2024-06-16 12:06:01.962 28088-28088 jdev.usbfileman          W  type=1400 audit(0.0:693912): avc: denied { read } for name="usb" dev="tmpfs" ino=17571 scontext=u:r:untrusted_app:s0:c202,c256,c512,c768 tcontext=u:object_r:usb_device:s0 tclass=dir permissive=0 app=com.github.mjdev.usbfileman
2024-06-16 12:06:01.962 28088-28088 jdev.usbfileman          W  type=1400 audit(0.0:693913): avc: denied { read } for name="/" dev="tmpfs" ino=10923 scontext=u:r:untrusted_app:s0:c202,c256,c512,c768 tcontext=u:object_r:device:s0 tclass=dir permissive=0 app=com.github.mjdev.usbfileman
2024-06-16 12:06:02.008 28088-28088 UsbMassStorageDevice     I  MAX LUN 0
2024-06-16 12:06:02.010 28088-28088 ScsiBlockDevice          D  inquiry response: ScsiInquiryResponse [peripheralQualifier=0, peripheralDeviceType=0, removableMedia=false, spcVersion=6, responseDataFormat=2]
2024-06-16 12:06:02.013 28088-28088 ScsiBlockDevice          I  Block size: 512
2024-06-16 12:06:02.013 28088-28088 ScsiBlockDevice          I  Last block address: 2000409263
2024-06-16 12:06:02.016 28088-28088 PartitionTableFactory    D  FileSystemPartitionTableCreator returned null
2024-06-16 12:06:02.017 28088-28088 GPT                      D  EFI test string ????????????????
2024-06-16 12:06:02.017 28088-28088 PartitionTableFactory    D  GPTCreator returned null
2024-06-16 12:06:02.018 28088-28088 PartitionTableFactory    D  Found partition table MasterBootRecordCreator
2024-06-16 12:06:03.240 28088-9622  jdev.usbfileman          I  This is non sticky GC, maxfree is 8388608 minfree is 524288
2024-06-16 12:06:04.270 28088-28088 MainActivity             D  Capacity: -1
2024-06-16 12:06:04.271 28088-28088 MainActivity             D  Occupied Space: 0
2024-06-16 12:06:04.271 28088-28088 MainActivity             D  Free Space: -1
2024-06-16 12:06:04.271 28088-28088 MainActivity             D  Chunk size: 512
2024-06-16 12:06:04.424 28088-12569 MIUIScout App            W  Enter APP_SCOUT_WARNING State
2024-06-16 12:06:04.425 28088-12569 MIUIScout App            W   (Current message: duration=2592ms seq=1833 h=android.app.ActivityThread$H w=159)
2024-06-16 12:06:04.433 28088-12569 AppScoutStateMachine     D  dump trace dirName = com.github.mjdev.usbfileman-28088-APP_SCOUT_WARNING-2024-06-16-12-06-04
2024-06-16 12:06:04.448 28088-12968 MIUIScout App            W  Event:APP_SCOUT_WARNING Thread:main backtrace:
                                                                    at me.jahnen.libaums.libusbcommunication.LibusbCommunication.nativeBulkTransfer(Native Method)
                                                                    at me.jahnen.libaums.libusbcommunication.LibusbCommunication.bulkInTransfer(LibusbCommunication.kt:82)
                                                                    at me.jahnen.libaums.core.driver.scsi.ScsiBlockDevice.transferOneCommand(ScsiBlockDevice.kt:277)
                                                                    at me.jahnen.libaums.core.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:154)
                                                                    at me.jahnen.libaums.core.driver.scsi.ScsiBlockDevice.read(ScsiBlockDevice.kt:331)
                                                                    at me.jahnen.libaums.core.driver.ByteBlockDevice.read(ByteBlockDevice.kt:33)
                                                                    at me.jahnen.libaums.javafs.wrapper.device.FSBlockDeviceWrapper.read(FSBlockDeviceWrapper.java:59)
                                                                    at org.jnode.fs.exfat.DeviceAccess.getChar(DeviceAccess.java:95)
                                                                    at org.jnode.fs.exfat.UpcaseTable.toUpperCase(UpcaseTable.java:125)
                                                                    at org.jnode.fs.exfat.UpcaseTable.toUpperCase(UpcaseTable.java:133)
                                                                    at org.jnode.fs.exfat.NodeDirectory$VisitorImpl.foundNode(NodeDirectory.java:154)
                                                                    at org.jnode.fs.exfat.DirectoryParser.parseFile(DirectoryParser.java:291)
                                                                    at org.jnode.fs.exfat.DirectoryParser.parse(DirectoryParser.java:138)
                                                                    at org.jnode.fs.exfat.NodeDirectory.<init>(NodeDirectory.java:62)
                                                                    at org.jnode.fs.exfat.NodeDirectory.<init>(NodeDirectory.java:46)
                                                                    at org.jnode.fs.exfat.NodeEntry.getDirectory(NodeEntry.java:118)
                                                                    at me.jahnen.libaums.javafs.wrapper.fs.UsbFileWrapper.<init>(UsbFileWrapper.java:36)
                                                                    at me.jahnen.libaums.javafs.wrapper.fs.UsbFileWrapper.listFiles(UsbFileWrapper.java:136)
                                                                    at me.jahnen.libaums.core.usbfileman.UsbFileListAdapter.refresh(UsbFileListAdapter.kt:66)
                                                                    at me.jahnen.libaums.core.usbfileman.UsbFileListAdapter.<init>(UsbFileListAdapter.kt:108)
                                                                    at me.jahnen.libaums.core.usbfileman.MainActivity.setupDevice(MainActivity.kt:629)
                                                                    at me.jahnen.libaums.core.usbfileman.MainActivity.discoverDevice(MainActivity.kt:595)
                                                                    at me.jahnen.libaums.core.usbfileman.MainActivity.onCreate(MainActivity.kt:555)
                                                                    at android.app.Activity.performCreate(Activity.java:8522)
                                                                    at android.app.Activity.performCreate(Activity.java:8486)
[... cut ...]
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:223)
        at android.app.ActivityThread.main(ActivityThread.java:7664)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)
W/libEGL: EGLNativeWindowType 0xb40000758657a2e0 disconnect failed
```

### Code where problem occurs

grantfee, 2024-08-12:
Hi，Anybody can help me for this issue？ Some issues happened while supporting exfat file system on android

frank-yulingfeng, 2024-08-13:
libaums library only support fat32 file system.